doonny/PipeCNN #1: Configuration for pool_size and conv_stride

chrisliu54, 2017-05-27:
Since the *width* and *height* of a pooling rectangle could be *different*,  it seems [`pool_size`](https://github.com/doonny/PipeCNN/blob/master/project/host/main.cpp#L91) could not cover this sort of case.
So does it for [`conv_stride`](https://github.com/doonny/PipeCNN/blob/master/project/host/main.cpp#L89) if the stride size varies for *width* and *height*.

doonny, 2017-06-13:
Yes, for pooling, the width and height of the window must be the same.

Normally, most people use a square pooling/convolution window. However, for convolution, the stride does not need to be the same. You can add other stride parameters for the convolution kernel.

Currently, we does not intend to support this feature.

chrisliu54, 2017-06-16:
Well, [Keras](https://github.com/fchollet/keras/blob/master/keras/layers/pooling.py#L166) supports different length for width and height when configuring pooling windows. I think it would be better if you add this feature for general purpose.

doonny, 2017-06-18:
I see, could you give me an example of a specific nework that uses rectangle filter  window, something like AlexNet  ?   I would like to study the network structure and get an idea how such feature can be added. Since the hardware resource on FPGAs are always limited, there are always restrictions on the size of the filters. This is different from GPU/CPU opencl implementations.

closed 2017-07-30 by doonny