doonny/PipeCNN #3: Compile error on Intel FPGA SDK for OpenCL.

nilesh441, 2017-09-06:
This source is supposed to be compiled on Intel FPGA SDK for OpenCL but I am getting following error for arria 10. I am using following version of the Intel(R) FPGA SDK for OpenCL(TM), 64-Bit Offline Compiler.
**Version 17.0.2 Build 297**

Error:

> error: function 'read_channel_altera' is not supported by the Intel(R) FPGA SDK for OpenCL(TM), and no user definition is provided
> 
> error: function 'write_channel_altera' is not supported by the Intel(R) FPGA SDK for OpenCL(TM), and no user definition is provided

nilesh441, 2017-09-06:
Oops. It is Intel now. Resolved by replacing API name.

closed 2017-09-06 by nilesh441

doonny, 2017-09-07:
Currently, we recommand using OpenCL SDK 16.1 only.

nilesh441, 2017-09-08:
Thank you for the information. I appreciate your work on this. 
However I am able to compile it using OpenCL SDK 17. You just need to change **write_channel_altera** to **write_channel_intel** and **read_channel_altera** to **read_channel_intel** and then you go like surfer.

nilesh441, 2017-09-08:
I forgot mention that you need to use following macro at start in cl file.
`#pragma OPENCL_EXTENSION cl_intel_channels : enable`

rvakash, 2017-12-11:
Hi @nilesh441, 

I am trying to compile the project on Intel FPGA SDK for OpenCL. I changed FLOW = hw, in the Makefile. I havent changed anything else. It is working fine with the emulator. 

I am getting the following error - 
rvakash@build1:/data/05297/rvakash/PipeCNN1/PipeCNN/project$ make
aoc -v --report --profile -I device/RTL -L device/RTL -l rtl_lib.aoclib  -g device/conv_pipe.cl -o conv.aocx
Resolving library filenames to full paths
  lib_path = device/RTL
    lib_file = rtl_lib.aoclib
Resolved rtl_lib.aoclib to /data/05297/rvakash/PipeCNN1/PipeCNN/project/device/RTL/rtl_lib.aoclib
  lib_path = .
aoc: Environment checks are completed successfully.
You are now compiling the full flow!!
aoc: Selected default target board bdw_fpga_v1.0
aoc: Running OpenCL parser....
In file included from <built-in>:11140:
<command line>:2:30: warning: ISO C99 requires whitespace after the macro name
#define ACL_BOARD_bdw_fpga_v1.0 1
                             ^
<command line>:3:31: warning: ISO C99 requires whitespace after the macro name
#define AOCL_BOARD_bdw_fpga_v1.0 1
                              ^
2 warnings generated.
aoc: OpenCL parser completed successfully.
aoc: Compiling....
aoc: Linking with IP library ...
Compiler Warning: Kernel 'lrn': limiting to 3 concurrent work-groups to support variable work-group sizes. Specify a reqd_work_group_size if possible. Please see Known Issue #2 in the Release Definition document for more details.

+--------------------------------------------------------------------+
; Estimated Resource Usage Summary                                   ;
+----------------------------------------+---------------------------+
; Resource                               + Usage                     ;
+----------------------------------------+---------------------------+
; Logic utilization                      ;   68%                     ;
; ALUTs                                  ;   31%                     ;
; Dedicated logic registers              ;   38%                     ;
; Memory blocks                          ;   42%                     ;
; DSP blocks                             ;   15%                     ;
+----------------------------------------+---------------------------;
aoc: First stage compilation completed successfully.
Error: Compiler Error, not able to generate hardware

Makefile:135: recipe for target 'conv.aocx' failed
make: *** [conv.aocx] Error 1

Thank you

doonny, 2017-12-12:
It seems that the design could not be fitted on your device due to limited resources.
Please check with the quartus_sh.log file to see what happened.

BTW, what kind of board are you using ?

ACL_BOARD_bdw_fpga_v1.0 1 seems new to me

Fnajjar, 2018-04-23:
Hi @doonny @nilesh441 @rvakash,
Thnak you very much for your work, 
well i am actually tring to implement the PipeCNN on my Attila Arria 10 GX a reflex card with the intel openCL SDK v17, i have done the changed you've already menshend but still have somme errors and still can't generate my conv.aocx. help; what shall i do?

Here's the errors i get: 

In file included from /home/najjar/Documents/PipeCNN-master/project/__all_sources.cl:2:
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:69:24: error: Channel support is not enabled
channel channel_vec    data_ch    __attribute__((depth(0)));
                       ^
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:70:24: error: Channel support is not enabled
channel channel_vec    weight_ch  __attribute__((depth(0)));
                       ^
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:71:24: error: Channel support is not enabled
channel channel_scal   bias_ch    __attribute__((depth(8)));
                       ^
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:72:24: error: Channel support is not enabled
channel channel_scal   conv_ch    __attribute__((depth(CHN_DEPTH)));
                       ^
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:73:24: error: Channel support is not enabled
channel channel_scal   pool_ch    __attribute__((depth(CHN_DEPTH)));
                       ^
/home/najjar/Documents/PipeCNN-master/project/device/conv_pipe.cl:74:24: error: Channel support is not enabled
channel channel_scal   bypass_ch  __attribute__((depth(CHN_DEPTH)));

Thank you very much for your help :)
[image]

aazz44ss, 2018-04-23:
you have to enable channel
The code to enable channel is different between v16 and v17.
You can refer to “altera opencl best practice guide”

Fnajjar, 2018-04-23:
Hi @aazz44ss @doonny  @nilesh441 and @rvakash , well it worked successfully and it did generate the conv.aocx finally, but the problem is that i have an ERROR: CL_INVALID_BINARY,  Failed to create program with binary
Since, I am working on the OpenCL SDK v17.1, what shall I do to adapt it and make it work??
what have you done before? did you had the same error??

Thank you for the help :)

[image]

DasunBhanuka, 2018-04-23:
@Fnajjar @nilesh441 @rvakash @aazz44ss 

I've encountered the exact problem. This seems to be a common problem as I've seen several others also had the same problem can someone answer this?

```
Loading kernel/binary from file conv.aocx
ERROR: CL_INVALID_BINARY
Location: ../common/ocl_util.cpp:415
Failed to create program with binary
```
I can't seem to figure out what the problem is ?
I am running **OpenCL 16.1** on a **terasic DE1SoC** board.

aazz44ss, 2018-04-24:
this is because the aocx file doesn’t match your board.
You should check whether device 0 is de1soc board, or you compiled aocx file with correct BSP

hiratz, 2019-02-26:
> Hi @aazz44ss @doonny @nilesh441 and @rvakash , well it worked successfully and it did generate the conv.aocx finally, but the problem is that i have an ERROR: CL_INVALID_BINARY, Failed to create [...]

Hi, @Fnajjar , How many hours does it take for you to complete the compiling (generating the final .aocx file)?

nilesh441, 2019-02-26:
@hiratz, it generally takes 40-50 minutes for cyclone 5 SOC and 4-5 hours for arrria 10 FPGA board.

You are getting Invalid Binary error because it is not compiled for your device. Use the correct BSP and check your device again.

Right now I am not working on this but I remembered that it worked for me in first try.

hiratz, 2019-02-26:
> @hiratz, it generally takes 40-50 minutes for cyclone 5 SOC and 4-5 hours for arrria 10 FPGA board.
>
> You are getting Invalid Binary error because it is not compiled for your device. Use the correct [...]

@nilesh441 Thanks a lot. I'm also working on Intel Arria 10 GX FPGA with OpenCL. I guess here 4-5 hours is for full compilation instead of fast one, right? If so, the compiling speed is really fast considering such a large kernel size. I asked this question because my own kernel's compiling time has exceeded 40 hours, which is probably abnormal. So I just stopped it. My kernel has a similar resource utilization (ALUTs: 44%, FFs: 21%, RAMs 58%, DSP 4%) like the one shown by @rvakash , and its size is 1075 line code. I may need to review my kernel and something must be wrong.

I asked my friends who have OpenCL experience targeting FPGA, and knew that usually 10+ (< 20) hours for a full compilation should be a reasonable time.

BTW, have you tried the -fast-compile option and measured how much performance will drop if you apply a fast compilation? I think it is interesting to know such a comparison because it will tell us whether the performance drop from fast compilation is closely related to the specific design or is a relatively constant. The Intel document tells us it is "normally 10 to 20 percent reduction" with -fast-compile ("FPGA Programming with OpenCL*". I don't know whether it is allowed to put a link here, I only show the document's title.)

nilesh441, 2019-02-26:
@hiratz, my compilation was fast because I was doing it on machine with 300+ cores and 64GB ram. lower the ram, slower the compilation.

hiratz, 2019-02-26:
> @hiratz, my compilation was fast because I was doing it on machine with 300+ cores and 64GB ram. lower the ram, slower the compilation.

@nilesh441 Yep, the compilation occupies a large amount of memory.

sergio14890, 2020-04-16:
Hey guys!
I have the de1-soc board. 
Im try do make, but i have this error:

> Compiler Command: aoc -v -report -seed=3 -I device/RTL -L device/RTL -l rtl_lib.aoclib -board-package=C:\intelFPGA\17.1\hld\board\de1_soc -I C:\intelFPGA\17.1\hld/include/kernel_headers -g device/conv_pipe.cl -o conv.aocx
c:/Users/sergi/Downloads/PipeCNN-master/project/device/conv_pipe.cl:1094:16: error: attribute takes one argument
__attribute__((max_work_group_size(1,1,1))) // (x,y,z)
               ^
1 error generated.
> 

Do you know also about this?
>