doonny/PipeCNN #4: theoretical exaplanation of Host code.

nilesh441, 2017-09-11:
I am trying to understand the host and fpga code as well. However I am not able to link it with available sources, I have.
Can you please provide or suggest me and other viewer on this topic? Can you provide any link based on  which you have written your code?

doonny, 2017-09-15:
Can you provide the logfiles for the compilation ?

nilesh441, 2017-09-15:
I have attached all three log files, I found. I changed their extension to **.txt** as github does not allow **.log**. I am using Arria 10 GX development kit and OpenCL SDK 17 and compiling it on Quartus Pro 17.  

I have added **-lrt** as a linking option in Makefile to compile the host code. I have also changed some api name to match those with OpenCL SDK 17.  
e.g **read_channel_altera** to **read_channel_intel**.

For more information, see [https://github.com/doonny/PipeCNN/issues/3](url)

[conv_pipe.txt](https://github.com/doonny/PipeCNN/files/1305309/conv_pipe.txt)
[host_compile.txt](https://github.com/doonny/PipeCNN/files/1305310/host_compile.txt)
[quartus_sh_compile.txt](https://github.com/doonny/PipeCNN/files/1305311/quartus_sh_compile.txt)

doonny, 2017-09-17:
I did not see any errors in any of your log file, could you be more specific ?

BTW, if you are using SoC FPGAs, you need to change the arch from x86 to arm32 in the original Make file.

Again, we have not tried 17.0, so no guarantee the new version would work without changing the host or device code.

nilesh441, 2017-09-19:
Yeah there is no error in log file and I am able to run the program successfully and it gives the check **pass output**.
My question was how convolution layer is performed in your code. I am following following tutorial to understand the code.
[https://sushscience.wordpress.com/2016/12/04/understanding-alexnet/](url)
But I think you are following different approach by using the concepts of work-group and work-item.
Can you share some information on that stuff like how the calculation is divided into work-group and work-item? How data_init array is prepared?

doonny, 2017-09-20:
Please refer to our research paper on arXiv.

nilesh441, 2017-09-26:
I read your research paper but I am still not getting the concept behind the LANE_NUM and VEC_SIZE. Can you suggest the some link or pages that explains all these stuff.

doonny, 2017-09-27:
VEC_SIZE is used to control how many int8 data are grouped into one vectorized data.
LANE_NUM is used to define how many parallel convolution pipelines are used.
Increasing both values results improved throughput.

doonny, 2017-10-09:
@nilesh441 Please close this issue if there are no further quesions.

closed 2017-10-10 by nilesh441