doonny/PipeCNN #7: MaxPool Problem

AndrewSivrit, 2017-09-15:
I'm sorry, I'm too much hurry with this issue.
This problem is actual if add padding(=1) at right side and lower side. 
for geting result with sizes : size_x = 13, size_y = 13
More information:
input: size_x = 13, size_y = 13, pool_size = 2, pool_stride = 1,  depth = 512
result: size_x = 13, size_y = 13, depth = 512

doonny, 2017-09-15:
Hi @AndrewSivrit if you are using pool_stride=1, regardless what pool_size you are using, you will always get the same output size after pooling, it is the nature of the algorithm, not determined by the code.

So what kind of pooling operation you want to implement ?

AndrewSivrit, 2017-09-26:
Doony are you sure ? 
If I have feature map 13x13 and pool_size = 2, pool_stride = 1, I get result (13-1)x(13-1) feature map, because in the line (size = 13) filter of pooling have 12 unique position.

closed 2018-02-12 by AndrewSivrit

hammadj, 2018-03-24:
@AndrewSivrit I am also trying to use feature map 13x13 with pool_size = 2 and pool_stride = 1, getting an error `incorrect setting of pooling input/output size for layer-6!!!` 

How did you fix this?