doonny/PipeCNN #8: Inference result not displayed.

gandautama, 2017-09-24:
HI how to print:
" The inference result is n02123045 tabby, tabby cat   (the prob is 56.00) ." at last in your documentation? 
Thanks in advance..
Regards,
Ganda.

nilesh441, 2017-09-28:
I also faced this problem but I think latest commit contains this change.

gandautama, 2017-09-28:
Ok thanks i will close this issue.

doonny, 2017-10-09:
@gandautama Please close this issue if it is correctly fixed.

gandautama, 2017-10-09:
Thanks, I have pulled the last commit build it with no error. Sorry for the delay.

closed 2017-10-09 by gandautama

gandautama, 2017-10-13:
I forked and put the result for DE-10 NANO using OpenCL-17. [the output is here](https://github.com/gandautama/PipeCNN/blob/master/eclipse/pipe_2_0/out.txt).

doonny, 2017-10-14:
@gandautama Great THX. Which hw configurations are you using ?

gandautama, 2017-10-14:
I am using FPGA Board configuration (not simulation), [ here is the hardware param](https://github.com/gandautama/PipeCNN/blob/master/eclipse/pipe_2_0/device/hw_param.cl).