doonny/PipeCNN #11: Can't find arm32  lib when make host for de1soc

thinkoco, 2017-10-12:
When compiling de1soc  host ,can't find arm lib. Maybe ,modify Makefile Line 73 74 
 $(shell aocl compile-config)  --> $(shell aocl compile-config --arm)
 $(shell aocl link-config) --> $(shell aocl link-config--arm)
windows64,Quartus 16.1 and compile with SoC EDS 16.1 Command Shell

doonny, 2017-10-12:
please change PLATFORM = x86 in the makefile to arm32 before compile for arm platform.
the instructions are provided in the docs.

thinkoco, 2017-10-12:
yes , I had changed. I sets 2 paths (%ALTERAOCLSDKROOT%\bin  and %ALTERAOCLSDKROOT%\host\windows64\bin ) in environment Path the same as terasic DE1SOC opencl manual.  In SoC EDS 16.1 Command Shell “ aocl link-config --arm” can find arm lib path ,‘’ aocl link-config‘’ give x86 lib path （maybe for emulator）

1. aocl link-config --arm show：
`  -LD:\App\intelFPGA\16.1\hld\board\terasic\de1soc\arm32\lib -LD:/App/intelFPGA/16.1/hld/host/arm32/lib -Wl,--no-as-needed -lalteracl -lalterammdpcie -lelf -lstdc++`

2. aocl link-config  show：
`-LD:\App\intelFPGA\16.1\hld\board\terasic\de1soc\arm32\lib /libpath:D:/App/intelFPGA/16.1/hld/host/windows64/lib -lalterammdpcie alteracl.lib acl_emulator_kernel_rt.lib pkg_editor.lib libelf.lib acl_hostxml.lib`

thinkoco, 2017-10-12:
I See , When  compile host on DE1SOC，"aocl link-config" maybe correct.

closed 2017-10-12 by thinkoco