doonny/PipeCNN #12: make: aocl :命令未找到

tmxklqy, 2017-10-14:
Dear Prof. Wang,
    I install Xilinx's SDAccel, but it display make: aocl: 命令未找到。Can ocal application on SDAccel?

doonny, 2017-10-15:
you need to change VENDOR = altera to xilinx in the Makefile

tmxklqy, 2017-10-17:
think you very much. Can you tell me  how to use SDAccel to run the project, I import the project but it shows "WARN	: Could not find any SDSoC projects."

tmxklqy, 2017-10-19:
Dear Prof.Wang,
    I see in the last version, you said "Add support for Xilinx's SDAccel tool(2017.02)",  but I import you project it shows 
"make all 
make: *** 没有规则可以创建目标“all”。 停止。"
could you write a file which like a "READE.md" to introduce how to use SDAccel to run you project.

doonny, 2017-10-20:
@tmxklqy Simply type make. there is no make all. The make follow is the same for altera and xilinx.

tmxklqy, 2017-10-25:
thank you.
how to solve the problem: /usr/bin/ld: 找不到 -lxilinxopencl

tmxklqy, 2017-10-25:
what is "Xilinx's OpenCL SDK toolset", it is SDAccel?

doonny, 2017-11-06:
@tmxklqy Yes, it is SDAccel.

closed 2018-01-02 by doonny