doonny/PipeCNN #13: Build failing for Xilinx flow (hardware accelerator integration stage)

davenso, 2017-10-15:
I placed the line "#define XILINX" at the top of conv_pipe.cl
When I build, I get the following errors during hardware accelerator integration stage. Am I missing any step? 

```
INFO: [XOCC 60-251]   Hardware accelerator integration...

===>The following messages were generated while processing /PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.sim/sim_1/behav :
ERROR: [XOCC 10-426] cannot find port pool_ch15_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:736]
ERROR: [XOCC 10-426] cannot find port pool_ch15_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:735]
ERROR: [XOCC 10-426] cannot find port pool_ch15_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:734]
ERROR: [XOCC 10-426] cannot find port pool_ch14_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:733]
ERROR: [XOCC 10-426] cannot find port pool_ch14_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:732]
ERROR: [XOCC 10-426] cannot find port pool_ch14_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:731]
ERROR: [XOCC 10-426] cannot find port pool_ch13_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:730]
ERROR: [XOCC 10-426] cannot find port pool_ch13_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:729]
ERROR: [XOCC 10-426] cannot find port pool_ch13_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:728]
ERROR: [XOCC 10-426] cannot find port pool_ch12_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:727]
ERROR: [XOCC 10-426] cannot find port pool_ch12_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:726]
ERROR: [XOCC 10-426] cannot find port pool_ch12_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:725]
ERROR: [XOCC 10-426] cannot find port pool_ch11_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:724]
ERROR: [XOCC 10-426] cannot find port pool_ch11_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:723]
ERROR: [XOCC 10-426] cannot find port pool_ch11_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:722]
ERROR: [XOCC 10-426] cannot find port pool_ch10_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:721]
ERROR: [XOCC 10-426] cannot find port pool_ch10_TVALID on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:720]
ERROR: [XOCC 10-426] cannot find port pool_ch10_TDATA on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:719]
ERROR: [XOCC 10-426] cannot find port pool_ch9_TREADY on this module [/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/ipiprj/ipiprj.srcs/sources_1/bd/dr/ipshared/8aa7/hdl/verilog/maxPool.v:718]
ERROR: [XOCC 43-3322] Static elaboration of top level Verilog design unit(s) in library work failed
ERROR: [XOCC 60-399] vivado failed, please see log file for detail: '/PipeCNN/Emulation-HW/_xocc_link_pipecnn/impl/build/hw_em/pipecnn/sv/pipecnn_ipi/vivado.log'
ERROR: [XOCC 60-626] Kernel link failed to complete
ERROR: [XOCC 60-703] Failed to finish linking
make: *** [pipecnn.xclbin] Error 1

21:27:33 Build Finished (took 10m:11s.280ms)
```

doonny, 2017-10-16:
which version of SDx are you using ?

doonny, 2017-10-16:
Please IDE-based flow. For HW-emu, make-file-based flow is not supported current.

closed 2017-10-16 by davenso