doonny/PipeCNN #15: Xilinx flow

davenso, 2017-10-17:
When #define XILINX is set, generates this error.

```
‘write_event’ was not declared in this scope
                             0 /* flags, 0 means from host*/,0, NULL,&write_event[i]);
                                                                      ^~~~~~~~~~~
../src/host/main.cpp:466:71: error: ‘write_event’ was not declared in this scope
                              0 /* flags, 0 means from host*/,0, NULL,&write_event[i]);
                                                                       ^~~~~~~~~~~
../src/host/main.cpp: In function ‘int prepare()’:
../src/host/main.cpp:1414:5: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
     else
     ^~~~
../src/host/main.cpp:1418:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
  for(unsigned n = 0; n<layer_config[0][data_n]/VEC_SIZE; n++){
  ^~~
make: *** [src/host/main.o] Error 1
```

davenso, 2017-10-17:
Ok, I got this resolved. But when executing the program I get a different error:

```
ERROR: ERROR: Memory bank specified for kernel instance "memRead_1" of kernel "memRead" for argument index 21 does not match the physical connectivity from the binary.
Bank specified on host side is "M01_AXI" while bank from the binary is "M00_AXI".

ERROR: clSetKernelArg() for kernel "memRead", argument index 21.
```

xuke225, 2017-10-17:
@davenso  did you open #define USE_SDX_4DDR ? If you open this you should open Makefile as follow:
#--xp misc:map_connect=add.kernel.memRead_1.M_AXI_GMEM0.core.OCL_REGION_0.M00_AXI\
#--xp misc:map_connect=add.kernel.memRead_1.M_AXI_GMEM1.core.OCL_REGION_0.M01_AXI\
#--xp misc:map_connect=add.kernel.memRead_1.M_AXI_GMEM2.core.OCL_REGION_0.M02_AXI\
#--xp misc:map_connect=add.kernel.memWrite_1.M_AXI_GMEM0.core.OCL_REGION_0.M00_AXI\
#--xp misc:map_connect=add.kernel.lrn_1.M_AXI_GMEM.core.OCL_REGION_0.M00_AXI
#--max_memory_ports lrn\
#--xp misc:map_connect=add.kernel.lrn_1.M_AXI_GMEM0.core.OCL_REGION_0.M00_AXI\
#--xp misc:map_connect=add.kernel.lrn_1.M_AXI_GMEM1.core.OCL_REGION_0.M01_AXI

davenso, 2017-10-24:
I am not using a Makefile. I use SDx.

shrivinayak, 2017-12-10:
Any update on this one? 
I am facing the same error.

doonny, 2018-02-07:
Please try the latest updates.

shrivinayak, 2018-02-07:
Yes it worked for me. thanks!

closed 2018-02-07 by doonny