doonny/PipeCNN #17: Xilinx Flow - Run time hang

pragadeeshmcw, 2017-11-02:
Hello Prof. Wang,

I'm trying to run the PipeCNN on Alpha Data 7v3 FPGA. [xilinx_adm-pcie-7v3_1ddr_3_0]
I'm using SDx 2017.2 and software emulation runs properly giving correct results.
When hardware is built, the timing is not met for some paths and the tool reduces the clock speed to 170.3 MHz (from the original 200 MHz). 
But when i run the generated binary conv.xclbin on the FPGA it results in a hang at run time

```
Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel Pooling

Launching kernel MemWr with local size: 1, 1, 16  (global size: 27, 27, 96)

Launching kernel lrn with local size: 1, 1, 24  (global size: 27, 27, 24)
```

Could you please help me in figuring out the issue?
Thanks in advance

Regards

doonny, 2017-11-06:
There is a bug in xilinx's dma driver for Alpha Data 7v3's board. They might fix it in the futhure.
Currently, it is only tested on KCU1500 board.

pragadeeshmcw, 2017-11-06:
Dear Prof. Wang.
Thank you for taking your time in helping me with this issue

Regards

doonny, 2017-11-06:
@pragadeeshmcw you could also feed this issue back to AlphaData, and let them know many people have encountered this problem, they might fix it soon. otherwise, you may not never fix it at all.

closed 2017-11-06 by doonny