doonny/PipeCNN #18: Does it work on Windows 10?

moderato, 2017-11-05:
Hello. As the title said, does PipeCNN support running on Windows 10? Thanks!

doonny, 2017-11-06:
SDx does not support windows, but Intel's OpenCL SDK support windows. You can checkout with version of windows they support on Intel OpenCL SDK 's website.

moderato, 2017-11-19:
@doonny I have Intel's OpenCL SDK installed on Windows 10. The program will be running on an Arria 10. With this configuration is it possible to compile PipeCNN using mingw32 on Windows?

doonny, 2017-11-19:
I think aoc 17.1 support mingw64, but not SDx.
You'd better refer to official docs.

moderato, 2017-11-21:
I am using 17.0. I can't find if it supports mingw64 or not in the user guide, but I guess it's OK to use it. I got an error during the compilation but I'll open another issue and close this one. Thanks!

closed 2017-11-21 by moderato