doonny/PipeCNN #21: Compile error with alteracl.lib using mingw-w64 on Windows 10

moderato, 2017-11-21:
Hello! I am trying to compile the code with the Intel OpenCL FPGA SDK 17.0 and an Arria 10 board on Windows 10 using mingw-w64. I got an error when the makefile run a command looks like:

```bash
g++ ./host/main.o ../common/ocl_util.o ../common/timer.o -o run.exe -LC:/intelFPGA_pro/17.0/hld/board/a10_ref/windows64/lib -LC:/intelFPGA_pro/17.0/hld/host/windows64/lib -laltera_a10_ref_mmd -lalteracl -lacl_emulator_kernel_rt -lpkg_editor -llibelf -lacl_hostxml
```

and I got an error saying like:

```bash
C:/intelFPGA_pro/17.0/hld/host/windows64/lib/alteracl.lib(d:/SJ/nightly/17.0/290/w64/acds/hld/obj/windows64/acl/acl_program.obj).text[l_build_from_source_in_dir]+0xa2): undefined reference to `__imp__wassert'
C:/intelFPGA_pro/17.0/hld/host/windows64/lib/alteracl.lib(d:/SJ/nightly/17.0/290/w64/acds/hld/obj/windows64/acl/acl_program.obj).text[l_load_binary_pkg]+0xb36): undefined reference to `__security_check_cookie'
C:/intelFPGA_pro/17.0/hld/host/windows64/lib/alteracl.lib(d:/SJ/nightly/17.0/290/w64/acds/hld/obj/windows64/acl/acl_program.obj).xdata[$unwind$l_compute_hash]+0x10): undefined reference to `__GSHandlerCheck'
```
(too long and get truncated; only repeating these 3 errors.)

Anyone has an idea? Thanks in advance!!

thinkoco, 2017-11-28:
@moderato see issue  #11，It is recommended to use soc_eds_command_shell.bat

moderato, 2017-11-28:
Thanks! I solved it by following the instruction here: (The blog is written in Chinese) https://zhoujianshi.github.io/articles/2017/MinGW%20GCC%E4%BD%BF%E7%94%A8VC++%E7%9A%84%E9%9D%99%E6%80%81%E9%93%BE%E6%8E%A5%E5%BA%93/index.html

Basically it says: do the following for each .lib file you wanna link
```bash
reimp xxx.lib
dlltool -l xxx.a -d xxx.def -A -k
```
You can get the reimp application from this link if you're using mingw-w64: http://www.mingw.org/wiki/msvc_and_mingw_dlls

closed 2017-11-28 by moderato