doonny/PipeCNN #24: Compiler Error: Unrecognized function call: mult_add_fix8bx4

rvakash, 2017-11-27:
Hi,
I am using Intel SDK for OpenCL on Arria10 to run the PipeCNN - Alex net. I am getting the above error when I compile the kernel with this command - 
$ aoc device/conv_pipe.cl -o bin_fpga/conv_pipe.aocx  --board bdw_fpga_v1.0 -v -g
aoc: Environment checks are completed successfully.
You are now compiling the full flow!!
aoc: Selected target board bdw_fpga_v1.0
aoc: Running OpenCL parser....
In file included from <built-in>:11140:
<command line>:2:30: warning: ISO C99 requires whitespace after the macro name
#define ACL_BOARD_bdw_fpga_v1.0 1
                             ^
<command line>:3:31: warning: ISO C99 requires whitespace after the macro name
#define AOCL_BOARD_bdw_fpga_v1.0 1
                              ^
2 warnings generated.
aoc: OpenCL parser completed successfully.
aoc: Compiling....
Compiler Error: Unrecognized function call: mult_add_fix8bx4
Error: Optimizer FAILED.
Refer to conv_pipe/conv_pipe.log for details.

When I run the same on an emulator it is running fine and gives the expected output. 

Why is it not able to recognise the function mult_add_fix8bx4 ? Should it be compiled separately ?

Thanks,
Akash

thinkoco, 2017-11-28:
@rvakash Yes,it should be make first in the IP core file.

rvakash, 2017-11-28:
@thinkoco There is no Makefile inside the IP core. Inside the RTL folder there is a Makefile which I have compiled and generated the rtl_lib.aoclib and rtl_lib.aoco files.

doonny, 2017-12-05:
@rvakash please use the makefile provided. Your compile command is not correct.

rvakash, 2017-12-10:
Hi, 
Thanks, I was able to run it on the emulator with the Makefile provided. 
But with fpga I am getting -
aoc: First stage compilation completed successfully.
Error: Compiler Error, not able to generate hardware

I have changed FLOW = hw in Makefile. Anything else I need to change ?

rvakash, 2017-12-11:
I am getting the following error -

rvakash@build1:/data/05297/rvakash/PipeCNN1/PipeCNN/project$ make
aoc -v --report --profile -I device/RTL -L device/RTL -l rtl_lib.aoclib -g device/conv_pipe.cl -o conv.aocx
Resolving library filenames to full paths
lib_path = device/RTL
lib_file = rtl_lib.aoclib
Resolved rtl_lib.aoclib to /data/05297/rvakash/PipeCNN1/PipeCNN/project/device/RTL/rtl_lib.aoclib
lib_path = .
aoc: Environment checks are completed successfully.
You are now compiling the full flow!!
aoc: Selected default target board bdw_fpga_v1.0
aoc: Running OpenCL parser....
In file included from :11140:
:2:30: warning: ISO C99 requires whitespace after the macro name
#define ACL_BOARD_bdw_fpga_v1.0 1
^
:3:31: warning: ISO C99 requires whitespace after the macro name
#define AOCL_BOARD_bdw_fpga_v1.0 1
^
2 warnings generated.
aoc: OpenCL parser completed successfully.
aoc: Compiling....
aoc: Linking with IP library ...
Compiler Warning: Kernel 'lrn': limiting to 3 concurrent work-groups to support variable work-group sizes. Specify a reqd_work_group_size if possible. Please see Known Issue #2 in the Release Definition document for more details.

+--------------------------------------------------------------------+
; Estimated Resource Usage Summary ;
+----------------------------------------+---------------------------+
; Resource + Usage ;
+----------------------------------------+---------------------------+
; Logic utilization ; 68% ;
; ALUTs ; 31% ;
; Dedicated logic registers ; 38% ;
; Memory blocks ; 42% ;
; DSP blocks ; 15% ;
+----------------------------------------+---------------------------;
aoc: First stage compilation completed successfully.
Error: Compiler Error, not able to generate hardware

Makefile:135: recipe for target 'conv.aocx' failed
make: *** [conv.aocx] Error 1

Thank you

rvakash, 2017-12-12:
Hi, I was able to generate the conv.aocx file. When I run it I am getting a new error. 

pnirali@harp8:/data/05297/pnirali/PipeCNN2/project$ ./run.exe conv.aocx
./run.exe: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by ./run.exe)

doonny, 2017-12-12:
Please install required libs for OpenCL SDK. check it with intel's latest user manual.

LCF2016, 2017-12-29:
I need a license for both quartus and opencl,someone help me,please

closed 2018-01-02 by doonny

arun0303, 2018-07-20:
hi Prof Doony,
I am also having the same error while running the file in DE10 Standard kit
./run.exe conv.aocx
./run.exe: libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./run.exe)
I had installed all the required libs as mentioned in intel latest manual.
Kindly help me with this issue

Frankzd, 2019-03-19:
Hi @rvakash I am facing the same problem that you met before 
[image]

do you remember how to solve it?

I am using PAC_A10 and SDK 17.1 by the way.