doonny/PipeCNN #25: a problem about running the project in sdaccel gui

wjqwsp, 2017-11-28:
I put the image.dat,weights.dat and fc8.dat in the data folder, build and run the project.The build process finishes successfully.However, when I attempt to run the exe file,it finishes so quick and no errors occur.The console output is very short:

``` shell
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

61063552 total weights read 
154587 bytes image read 
```

I'm using SDaccel 2017.2 gui mode.Why is the output log so short?I don't see any output files created after I run the project.

wjqwsp, 2017-11-28:
I try the make compiling flow, and the output is similar.But this time I know it fails:
``` shell
[root@gitlabci src]# ./run.exe conv.xclbin 
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

61063552 total weights read 
154587 bytes image read 
Segmentation fault (core dumped)
```

How can I solve this?

wjqwsp, 2017-11-28:
I have fixed it.The issue will be closed.

closed 2017-12-05 by doonny

sinatv52, 2019-07-23:
@wjqwsp  may you pls say what was the problem? thanks