doonny/PipeCNN #26: ERROR: CL_COMPILER_NOT_AVAILABLE

moderato, 2017-11-28:
Hello! I compile the project in emulator mode and get the following error when I run "run.exe conv.aocx" with alexnet data:

```
run.exe conv.aocx
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs
***************************************************

61063552 total weights read
154587 bytes image read
1024 total output reference read

Platform: Intel(R) FPGA SDK for OpenCL(TM)
Using 1 device(s)
  Device 0: EmulatorDevice : Emulated Device
Device OpenCL Version: OpenCL 1.0 Intel(R) FPGA SDK for OpenCL(TM), Version 17.0
Device Max Compute Units: 1
Device Max WorkGroup Size: 2147483647
Device Max WorkItem Size: 2147483647
Device Global Memory Size: 2048 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 1000 Mhz

Loading kernel/binary from file conv.aocx
ERROR: CL_COMPILER_NOT_AVAILABLE
Location: ../common/ocl_util.cpp:429
Failed to build program with source
```

Environment: Windows 10, MinGW-w64, Arria 10 board, Intel OpenCL SDK for FPGA 17.0, MSVC 12.0

Anyone got an idea? Thanks in advance!!

davenso, 2017-11-28:
Hi

The error shows that you are trying to "build 'OpenCL program' from source". But you already have a built program from source which was compiled offline when you ran aoc. Therefore, I believe in your host code you will need to "create 'OpenCL program' with binary".  checkout the OpenCL doc for clCreateProgramWithSource, clCreateProgramWithBinary and clBuildProgram.

moderato, 2017-11-29:
@davenso I compiled it in sw-emu mode. Looks like in ocl_util.cpp, if FPGA_DEVICE is defined the program will go clCreateProgramWithBinary otherwise it'll go clCreateProgramWithSource. Is the emulator regarded as one of the FPGA devices?

moderato, 2017-11-30:
@davenso I actually figured out what to do. The program code has no problem. When the make is completed, it tells me to run
```bash
env CL_CONTEXT_EMULATOR_DEVICE_ALTERA=n ./run.exe conv.aocx
```
instead of simply
```bash
./run.exe conv.aocx
```
where n is the number of devices you wanna run the program on. I set it to 1 and that solves the problem.

@doonny I'm not sure if this is a general case. Can you please check that out? If it is a general case the user instructions could be made clearer by updating the run command.

Many thanks to both of you!

doonny, 2017-12-05:
There are multiple scripts in the project folder like this:
setup_aoc_emu.sh
setup_aoc_hw.sh
... ...
you need to source one of these script to setup the correct enviroment first.
please refer to Altera's opencl guide for the correct usage of the design flow.

closed 2018-01-02 by doonny

ArjunaDeSoysa, 2018-03-18:
@davenso , i am using DE1-SoC and when i tried to run it i get this error;

```
./run.exe conv.aocx
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs
***************************************************
Platform: Altera SDK for OpenCL                                                           
Using 1 device(s)                                                                         
  Device 0: de1soc_sharedonlyCyclone V SoC Development Kit                                
Device OpenCL Version: OpenCL 1.0 Altera SDK for OpenCL, Version 16.0                     
Device Max Compute Units: 1                                                               
Device Max WorkGroup Size: 2147483647                                                     
Device Max WorkItem Size: 2147483647                                                      
Device Global Memory Size: 512 MBytes                                                     
Device Local Memory Size: 16 KBytes                                                       
Device Max Clock Freq: 1000 Mhz 

Loading kernel/binary from file conv.aocx 
ERROR: CL_INVALID_BINARY
Location: ../common/ocl_util.cpp:415
Failed to create program with binary
```

and there's not appear in this part after the `PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs`

```
61063552 total weights read 
154587 bytes image read 
1024 total output reference read 
```
Can you please help me to solve this error?

jiangdiankun, 2018-03-18:
I think you should to  run  aocl program /dev/acl0  conv.aocx  before ./run.exe conv.aocx

ArjunaDeSoysa, 2018-03-18:
@davenso, i alrady try it. And it say reprogramming successfully.but after
that. /run.exe conv.aocx, the same error occurred.

ArjunaDeSoysa, 2018-03-18:
@davenso i only changed in code is, in /project/host/layer_config.h , line 393:

```
char precision_config[][3] ={{8,  0, -4},//Layer-1
							{ 8,  0, -2},//Layer-2
							{ 8,  0, -1},//Layer-3
							{ 8, -1, -1},//Layer-4
							{ 8, -1, -1},//Layer-5
							{11, -1,  0},//Layer-6
							{10,  0,  2},//Layer-7
							{10,  2,  2}//Layer-8
							};
```
to

```
signed char precision_config[][3] ={{8,  0, -4},//Layer-1
							{ 8,  0, -2},//Layer-2
							{ 8,  0, -1},//Layer-3
							{ 8, -1, -1},//Layer-4
							{ 8, -1, -1},//Layer-5
							{11, -1,  0},//Layer-6
							{10,  0,  2},//Layer-7
							{10,  2,  2}//Layer-8
							};
```
because i get the error.

other than that i did all as in the user instruction.