doonny/PipeCNN #27: vgg16 results wrong

wjqwsp, 2017-11-30:
The default network in the code is AlexNet, and I have successfully built and run it in xilinx kcu1500  board.However,when I modified some code and attempted to change the network to vgg16,I built the project successfully but got wrong results.I am a beginner in CNN and opencl.I think I really need some guide on how to change the code from the default AlexNet to vgg16.I can't see any documents in the readme or the user instruction.

　　This is the AlexNet result which seems correct:
``` 
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

61063552 total weights read 
154587 bytes image read 
1024 total output reference read 

Platform: Xilinx
Using 1 device(s)
  Device 0: xilinx:kcu1500:4ddr-xpr:4.0
Device OpenCL Version: OpenCL 1.0
Device Max Compute Units: 1
Device Max WorkGroup Size: 4096
Device Max WorkItem Size: 4096
Device Global Memory Size: 16384 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 500 Mhz

Loading kernel/binary from file conv.xclbin
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy

Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel Pooling

Launching kernel MemWr with local size: 1, 1, 16  (global size: 27, 27, 96)

Launching kernel lrn with local size: 1, 1, 24  (global size: 27, 27, 24)

Executing Layer 2:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel Pooling

Launching kernel MemWr with local size: 1, 1, 16  (global size: 13, 13, 256)

Launching kernel lrn with local size: 1, 1, 64  (global size: 13, 13, 64)

Executing Layer 3:

Launching single work-item kernel winbuffer

[... 410 lines cut ...]
    MemWr: 13.308 ms
    Lrn  : 0.000 ms
  Layer-16:
    MemRd: 18446744061093.504 ms
    Conv : 18446744061093.090 ms
    Pool : 0.000 ms
    MemWr: 3.383 ms
    Lrn  : 0.000 ms

Total kernel runtime 55340232221128.656 ms 
Batch size = 1, average process time per batch: 55340232221128.656 ms 

Start verifying results ...
Selected item = 0 from the combined batch results in fc buffers
Item=0 is wrong (result=-3.000000, golden_ref=-6.000000)
Item=1 is wrong (result=0.000000, golden_ref=3.000000)
Item=2 is wrong (result=-4.000000, golden_ref=-8.000000)
Item=3 is wrong (result=-4.000000, golden_ref=-9.000000)
Item=4 is wrong (result=-1.000000, golden_ref=-5.000000)
Item=5 is wrong (result=-4.000000, golden_ref=-1.000000)
Item=6 is wrong (result=-3.000000, golden_ref=-12.000000)
Item=7 is wrong (result=2.000000, golden_ref=7.000000)
Item=8 is wrong (result=-1.000000, golden_ref=10.000000)
Totally 946 Wrong Results
```

wjqwsp, 2017-12-05:
In CPU-Emulation mode, vgg16 will crash:
```
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

138455872 total weights read 
150528 bytes image read 
1024 total output reference read 

Linux:3.10.0-514.26.2.el7.x86_64:#1 SMP Tue Jul 4 15:04:05 UTC 2017:x86_64
--- 
XILINX_OPENCL=""
LD_LIBRARY_PATH="/root/xil-kcu1500/xbinst/runtime/lib/x86_64:/var/opt/xilinx/SDx/2017.2/runtime/lib/x86_64:/var/opt/xilinx/SDx/2017.2/lib/lnx64.o:/root/xil-kcu1500/xbinst/runtime/lib/x86_64:"
--- 

Platform: Xilinx
Using 1 device(s)
  Device 0: xilinx_kcu1500_4ddr-xpr_4_0
Device OpenCL Version: OpenCL 1.0
Device Max Compute Units: 1
Device Max WorkGroup Size: 4096
Device Max WorkItem Size: 4096
Device Global Memory Size: 16384 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 200 Mhz

Loading kernel/binary from file conv.xclbin
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy
WARNING: unaligned host pointer detected, this leads to extra memcpy

Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching kernel MemWr with local size: 1, 1, 16  (global size: 224, 224, 64)

[... 88 lines cut ...]
7fa2ea3e0000-7fa2ea3f7000 r-xp 00000000 fd:00 68890931                   /usr/lib64/libpthread-2.17.so
7fa2ea3f7000-7fa2ea5f6000 ---p 00017000 fd:00 68890931                   /usr/lib64/libpthread-2.17.so
7fa2ea5f6000-7fa2ea5f7000 r--p 00016000 fd:00 68890931                   /usr/lib64/libpthread-2.17.so
7fa2ea5f7000-7fa2ea5f8000 rw-p 00017000 fd:00 68890931                   /usr/lib64/libpthread-2.17.so
7fa2ea5f8000-7fa2ea5fc000 rw-p 00000000 00:00 0 
7fa2ea5fc000-7fa2ea5fe000 r-xp 00000000 fd:00 67159275                   /usr/lib64/libdl-2.17.so
7fa2ea5fe000-7fa2ea7fe000 ---p 00002000 fd:00 67159275                   /usr/lib64/libdl-2.17.so
7fa2ea7fe000-7fa2ea7ff000 r--p 00002000 fd:00 67159275                   /usr/lib64/libdl-2.17.so
7fa2ea7ff000-7fa2ea800000 rw-p 00003000 fd:00 67159275                   /usr/lib64/libdl-2.17.so
7fa2ea800000-7fa2ea8e9000 r-xp 00000000 fd:03 64424729586                /var/opt/xilinx/SDx/2017.2/lib/lnx64.o/libprotobuf.so.7
7fa2ea8e9000-7fa2eaae9000 ---p 000e9000 fd:03 64424729586                /var/opt/xilinx/SDx/2017.2/lib/lnx64.o/libprotobuf.so.7
7fa2eaae9000-7fa2eaaee000 rw-p 000e9000 fd:03 64424729586                /var/opt/xilinx/SDx/2017.2/lib/lnx64.o/libprotobuf.so.7
7fa2eaaee000-7fa2eab0f000 r-xp 00000000 fd:00 68656404                   /usr/lib64/ld-2.17.so
7fa2eab11000-7fa2eaba2000 rw-p 00000000 00:00 0 
7fa2eabeb000-7fa2eacf1000 rw-p 00000000 00:00 0 
7fa2ead0d000-7fa2ead0f000 rw-p 00000000 00:00 0 
7fa2ead0f000-7fa2ead10000 r--p 00021000 fd:00 68656404                   /usr/lib64/ld-2.17.so
7fa2ead10000-7fa2ead11000 rw-p 00022000 fd:00 68656404                   /usr/lib64/ld-2.17.so
7fa2ead11000-7fa2ead12000 rw-p 00000000 00:00 0 
7fffd9585000-7fffd95a7000 rw-p 00000000 00:00 0                          [stack]
7fffd95a8000-7fffd95aa000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
CPU emulation compute unit exited unexpectedly
Terminated
```

doonny, 2018-02-07:
Please try the latest updates.

closed 2018-02-13 by doonny