doonny/PipeCNN #28: Logic Optimization failing when compiling on AWS F1 for xilinx_aws-vu9p-f1_4ddr-xpr-2pr_4_0

shrivinayak, 2017-12-02:
[Finished 1st of 5 tasks (FPGA synthesis). Elapsed time: 00h 49m 48s.
.....
Finished 2nd of 5 tasks (FPGA logic optimization). Elapsed time: 00h 06m 40s.
....
ERROR: [XOCC 60-704] Integration error, problem implementing OCL region, place_design ERROR
ERROR: [XOCC 60-626] Kernel link failed to complete
ERROR: [XOCC 60-703] Failed to finish linking
make: *** [binary_container_1.xclbin] Error 1](url)

>> Integration error, problem implementing OCL region. 

Is this something I am doing wrong?

wjqwsp, 2017-12-03:
I have met a similar problem, which was solved when I ran the project in a more powerful computer.

doonny, 2017-12-05:
You need more memory. 32GB is prefered as told by Xilinx.

shrivinayak, 2017-12-06:
Thanks,
I was able to compile on c4.4xlarge AWS

closed 2017-12-06 by shrivinayak