doonny/PipeCNN #29: BSP for DE1-soc

SmartRoof, 2017-12-08:
Hi sir,
According to the description, the version of  Intel's OpenCL SDK v16.1 is used in this project. 
We would like to test run the program on the Terasic De1-SoC board. 
But we found only the BSP for Altera SDK OpenCL 16.0 is provided in the official webpage.
Is it the one you used in this project?
And is it work fine in Intel's OpenCL SDK v16.1?

[image]

Thank you

thinkoco, 2017-12-08:
Yes,it's OK. It contains the same version of the driver in 16.0 and 16.1.Hardware template can be compiled  in  16.1 and runtime is ok

doonny, 2017-12-08:
v17.0 is also OK.

closed 2017-12-16 by SmartRoof