doonny/PipeCNN #33: Makefile:129: recipe for target 'run.exe' failed

LCF2016, 2017-12-30:
Hello! I compile the project  and get the following error 
lcf@lcf-9020:~/work/PipeCNN-master/project$ make
g++   ./host/main.o ../common/ocl_util.o ../common/timer.o -o run.exe -L/home/lcf/intelFPGA/16.1/hld/board/de10_standard/arm32/lib -L/home/lcf/intelFPGA/16.1/hld/host/arm32/lib -L/home/lcf/intelFPGA/16.1/hld/host/linux64/lib -Wl,--no-as-needed -lalteracl -lalterammdpcie -lstdc++ -lelf 
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/board/de10_standard/arm32/lib/libalteracl.so when searching for -lalteracl
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/host/arm32/lib/libalteracl.so when searching for -lalteracl
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/board/de10_standard/arm32/lib/libalterammdpcie.so when searching for -lalterammdpcie
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/host/arm32/lib/libalterammdpcie.so when searching for -lalterammdpcie
/usr/bin/ld: cannot find -lalterammdpcie
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/board/de10_standard/arm32/lib/libelf.so when searching for -lelf
/usr/bin/ld: skipping incompatible /home/lcf/intelFPGA/16.1/hld/host/arm32/lib/libelf.so when searching for -lelf
collect2: error: ld returned 1 exit status
Makefile:129: recipe for target 'run.exe' failed
make: *** [run.exe] Error 1

Anyone got an idea? Thanks in advance!!

LCF2016, 2017-12-30:
Hi,I have changed PLATFORM = arm32 and FLOW  = hw in the Makefile.
I am getting the following error
lcf@lcf-9020:~/work/PipeCNN-master/project$ make
arm-linux-gnueabihf-g++   ./host/main.o ../common/ocl_util.o ../common/timer.o -o run.exe -L/home/lcf/intelFPGA/16.1/hld/board/de10_standard/arm32/lib -L/home/lcf/intelFPGA/16.1/hld/host/arm32/lib -L/home/lcf/intelFPGA/16.1/hld/host/linux64/lib -Wl,--no-as-needed -lalteracl -lalterammdpcie -lstdc++ -lelf 
./host/main.o: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
Makefile:129: recipe for target 'run.exe' failed
make: *** [run.exe] Error 1

Thanks

LCF2016, 2017-12-30:
Hi,problem have solved,I have removed the .o file.but got a new problem.
Could not acquire a valid license for the Intel(R) FPGA SDK for OpenCL(TM).
Error: Verilog generator FAILED.
Refer to conv/conv_pipe.log for details.

Makefile:135: recipe for target 'conv.aocx' failed
make: *** [conv.aocx] Error 1

I don't have a  license,someone help me,please.
Thanks.

doonny, 2018-01-02:
Please contact with your board's supplier to accquire a license.

closed 2018-01-02 by doonny