doonny/PipeCNN #34: error: Channel support is not enabled

LCF2016, 2017-12-30:
Hi,I compiled project with Intel OpenCL SDK for FPGA 17.1 and get the following error
In file included from /home/wangjf/PipeCNN/project/__all_sources.cl:2:
PipeCNN/project/device/conv_pipe.cl:75:24: error: Channel support is not enabled
channel channel_vec    data_ch    __attribute__((depth(0)));

Anyone got an idea? Thanks in advance!!

LCF2016, 2017-12-31:
I changed OPENCL_EXTENSION to OPEN EXTENSION  which can solve the above problem.
thanks!

closed 2018-01-02 by doonny

bagumondigi, 2018-05-22:
I have solved the problem by following this thread in Altera forum: https://www.alteraforum.com/forum/showthread.php?t=57424
in conv_pipe.cl change to: #pragma OPENCL EXTENSION cl_intel_channels : enable

lihao2333, 2018-06-14:
for `aoc` with version 17.1, you should do two thing in `conv_pipe.cl`  
1. change OPENCL_EXTENSION to OPENCL EXTENSION
2. replace all `altera` into `intel` in this file