doonny/PipeCNN #37: HW Configuration for DE1SOC

SmartRoof, 2018-01-02:
Hi, referring to instructions,  the best result of AlexNet model executed on De1Soc is 149 ms. 
However, the best result i got is just around 450 ms with the following hw configurations: 

- VEC_SIZE = 8
- LANE_NUM = 4
- CONV_GP_SIZE_X = 7

I do tried to increase the LANE_NUM to 8. But i got the following error even-though the resources of the device is not fully used up.

"kernel cannot fit into device"

Could you kindly share us the appropriate hw configuration for the VEC_SIZE, LANE_NUM, and CONV_GP_SIZE_X??

Thank you

doonny, 2018-01-05:
Use the following configurations for DE1-SOC
VEC_SIZE = 8
LANE_NUM = 8
CONV_GP_SIZE_X = 7

the resource utilizations are like this:

Registers: 67,316
Logic utilization: 28,961 / 32,070 ( 90 % ) ( 90 % )
I/O pins: 103 / 457 ( 23 % )
DSP blocks: 73 / 87 ( 84 % )
Memory bits: 2,219,759 / 4,065,280 ( 55 % )
M10K blocks: 397 / 397 ( 100 % )
Actual clock freq: 122.529999912
Kernel fmax: 122.53
1x clock fmax: 122.53
2x clock fmax: 301.2
Highest non-global fanout: 3075

LCF2016, 2018-01-05:
Hi,@doonny could you kindly also share us the appropriate hw configuration of DE10_Standard board?
Thanks a lot.

johnnydept, 2018-01-05:
Hello prof @doonny , Why DE1-SOC have 32k Logic? Suppose it has 85k Logic element.

doonny, 2018-01-06:
@johnnydept Hi, the number is ALM not LE.
@LCF2016 the following configurations should work for DE10 std, but it is not the optimal one.
VEC_SIZE = 16
LANE_NUM = 8
CONV_GP_SIZE_X = 7

closed 2018-02-07 by doonny