doonny/PipeCNN #42: OpenCL runtime error

HeroGian, 2018-01-24:
Dear doonny,

I'm very interested in testing your PipeCNN in my Zynq Ultrascale ZCU102. 
I have compiled the source code with Xilinx SDSoC v2017.1 with _zcu102_es1_ocl_ platform, then before launching the PipeCNN I have sent these commands:

```
cd /mnt
cp libxilinxopencl.so /usr/lib
export XILINX_OPENCL=/mnt
```
(libxilinxopencl.so is the opencl library for aarch64), then for launching the CNN:

`./PipeCNN.elf conv.aocx`

and finally the output is:

```
61063552 total weights read
154587 bytes image read
1024 total output reference read

ERROR: No device found
ERROR: CL_DEVICE_NOT_FOUND
```

Could you give me some help?

Thanks in advance

doonny, 2018-01-25:
Hi, PipeCNN has not been tested by using SDSoC, it is better to use SDAccel.

HeroGian, 2018-01-26:
Thanks for the reply, I'll try to get a SDAccel license

closed 2018-01-28 by doonny