doonny/PipeCNN #45: collect2: error: ld returned 1 exit status Makefile:129: recipe for target 'run.exe' failed

ArjunaDeSoysa, 2018-02-09:


hammadj, 2018-02-09:
Try running the init_opencl.sh script first: `source ~/intelFPGA/16.1/hld/init_opencl.sh`

hammadj, 2018-02-09:
Also, if you're doing sw_emu you want PLATFORM = x86 (assuming your running sw_emu not on your computer and not the DE1 board) and also to change your AOCL_BOARD_PACKAGE_ROOT env var to one that has x86 support like s5_ref.

ArjunaDeSoysa, 2018-02-10:
@hammadj ,First one (run `source ~/intelFPGA/16.1/hld/init_opencl.sh`) , but the same problem occur.
and Env Var are;

`export QUARTUS_ROOTDIR=/home/arjuna/intelFPGA_lite/16.1/quartus
#export QUARTUS_ROOTDIR_OVERRIDE=/home/arjuna/intelFPGA/16.1/quartus
export ALTERAOCLSDKROOT=/home/arjuna/intelFPGA/16.1/hld
export PATH=$PATH:$QUARTUS_ROOTDIR/bin:$ALTERAOCLSDKROOT/linux64/bin:$ALTERAOCLSDKROOT/bin:/home/arjuna/intelFPGA/16.1/embedded/ds-5/bin
export LD_LIBRARY_PATH=$ALTERAOCLSDKROOT/linux64/lib
export AOCL_BOARD_PACKAGE_ROOT=$ALTERAOCLSDKROOT/board/terasic/de1soc
export QUARTUS_64BIT=1
export LM_LICENSE_FILE=/home/arjuna/Documents/Licence_altera/`

like this.
can i know what do you mean by 'change your AOCL_BOARD_PACKAGE_ROOT env var to one that has x86 support like s5_ref'

I tried to run under the “Embedded_Command_Shell.sh”,
but it also didn't work.please help me to solve this problem.

Thank you.

ArjunaDeSoysa, 2018-02-11:
Thank You @hammadj . Finaly i got it. it's worked after change the AOCL_BOARD_PACKAGE_ROOT env var to s5_ref.

closed 2018-02-11 by ArjunaDeSoysa

ArjunaDeSoysa, 2018-03-19:
@hammadj , i did as you said , change the AOCL_BOARD_PACKAGE_ROOT env var to s5_ref.
it ok. it will generated run.exe and conv.aocx.
But the problem is when it run ,the error occur,  
```
arjuna@arjuna-Inspiron-3542:~/Desktop/SW/PipeCNN-master/project$ CL_CONTEXT_EMULATOR_DEVICE_ALTERA=1 ./run.exe conv.aocx
./run.exe: error while loading shared libraries: libaltera_s5_ref_mmd.so: cannot open shared object file: No such file or directory
```
can you please tell me the how to solve this problem?

Thank You.

[image]

aazz44ss, 2018-03-19:
Can you execute 
aocl diagnose
and attach out put message

This message may because you have not set your board path correctly

ArjunaDeSoysa, 2018-03-19:
@aazz44ss, aocl diagnose is ;
```
--------------------------------------------------------------------
No board diagnose routine supplied.                               
Please consult your board manufacturer's documentation or support   
team for information on how to debug board installation problems.   
--------------------------------------------------------------------

```
Can you please help me to solve this?

aazz44ss, 2018-03-19:
I find you are using emulator, it should work without installing board.

Try
env CL_CONTEXT_EMULATOR_DEVICE_ALTERA=1 ./run.exe conv.aocx

ArjunaDeSoysa, 2018-03-19:
@aazz44ss , the same error occur;
```
arjuna@arjuna-Inspiron-3542:~/Desktop/SW/PipeCNN-master/project$ env CL_CONTEXT_EMULATOR_DEVICE_ALTERA=1 ./run.exe conv.aocx
./run.exe: error while loading shared libraries: libaltera_s5_ref_mmd.so: cannot open shared object file: No such file or directory
```
after `source /etc/profile` also i `unset AOCL_BOARD_PACKAGE_ROOT`

but it will Selected default target board as s5_ref.

```
 aoc: Environment checks are completed successfully.
You are now compiling the full flow!!
aoc: Selected default target board s5_ref
aoc: Running OpenCL parser....
aoc: OpenCL parser completed successfully.
aoc: Compiling for Emulation ....
aoc: Emulator Compilation completed successfully.
```

hammadj, 2018-03-19:
@ArjunaDeSoysa Make sure you run the init_opencl.sh script:

`source ~/intelFPGA/16.1/hld/init_opencl.sh`

Or whatever the path to the intel FPGA SDK is

Meyerjan, 2018-03-26:
hi, i have the same error
arjuna@arjuna-Inspiron-3542:~$ aocl diagnose
--------------------------------------------------------------------
No board diagnose routine supplied.                               
Please consult your board manufacturer's documentation or support   
team for information on how to debug board installation problems.   
--------------------------------------------------------------------
Are you solve this?

ArjunaDeSoysa, 2018-03-26:
@Meyerjan  it seems like you try it on host computer. you want try it on FPGA board that you use. after entered the your board environment.