doonny/PipeCNN #46: Error: ip-generate FAILED. make: *** [conv.aocx] Error 1

ArjunaDeSoysa, 2018-02-11:
Dear Prof. Wang,

I use De1-SoC and I changed,

    VEC_SIZE  = 8
    LANE_NUM = 8 
    CONV_GP_SIZE_X = 7  , as the user instruction.
and 
    PLATFORM = arm32
    FLOW        = hw

After, aoc: First stage compilation completed successful,
I got this error,

`Error: qsys-generate failed with exit code 1: 35 Errors, 42 Warnings`
`Error: ip-generate FAILED.`
`Refer to conv/conv_pipe.log for details.`

`Makefile:139: recipe for target 'conv.aocx' failed`
`make: *** [conv.aocx] Error 1`

can you please suggest what should I do? 

Thank You.

doonny, 2018-02-12:
you could also use a smaller value of VEC_SIZE

ArjunaDeSoysa, 2018-02-12:
@doonny, I mean except reduce the value of  LANE_NUM and VEC_SIZE , Is there any other ways to reduce Logic utilization?
because I need to get **lowest Excution Time**.(by reduce LANE_NUM=4, i got  just around **450 ms**  )

Thank You.

doonny, 2018-02-13:
Please do not use the '--profile' option when you compile the kernel. That will cost you a lot of LUTs.
You should get the same results as we do.

ArjunaDeSoysa, 2018-02-13:
Ok. I got it. now its complete in 149ms. Thank You @doonny .

closed 2018-02-13 by ArjunaDeSoysa