doonny/PipeCNN #47: Error message by SDAccel 2017.4

nslmike, 2018-02-13:
1. It is needed to rename platform to xilinx_kcu1500_dynamic_5_0 for SDAccel 2017.4

2. A lot of warnings like this "device/conv_pipe_xilinx.cl:680:708: warning: double precision constant requires cl_khr_fp64, casting to single precision"

3. Finally an error message:
ERROR: [XOCC 60-896] For unified platforms, please use -c or -l
ERROR: [XOCC 60-598] Kernel build setup failed to complete
ERROR: [XOCC 60-702] Failed to finish compilation and linking
Makefile:142: recipe for target 'conv.xclbin' failed
make: *** [conv.xclbin] Error 1

nslmike, 2018-02-13:
I've added '-c' option before '-o' and it works.

doonny, 2018-02-13:
Great.
However, 2017.4 generates bad results, so we do not suggests to use this version.

closed 2018-02-13 by doonny

mattpcaswell, 2018-11-16:
Did you ever find a solution to that error @nslmike?

monsterhermanmonster1, 2019-06-12:
Same here - fixed the LRN compile error that apparently happens with 2018.3 (and .2 as well).
But when executing it throws that exact same error as nslmike posted above.

_(First thought thats due to WORK_GROUP_SIZE being reported as -1 by the RTE, but verified that -1 is the case reported in all my other OpenCL apps too. So think can safely rule that out.)_

Changed the VECSIZE=8 and LANES=8, recompiled (for VCU1525, VU9P), fine, but at runtime same issue. Local size then (1,1,12) of course because 96/8.

Anybody a clue how to debug that ? My suspicion is that the RTE (XRT driver) somehow gets confused...

BR !

P.S. for fixing that llvm based PHInode error code when compiling: It's in that first for-loop in the LRN kernel. Simply dont direct assign from global to local buffer along with that heftily looking index-arithmetics, but separate that one for loop into two and copy from global to a private buffer in the first (for as many few iterations as VECSIZE is large) and then, in the subsequent for loop, copy from that private to the local buffer. presto bob's ur uncle.

monsterhermanmonster1, 2019-06-12:
on VCU1525 (VU9P) and SDAccel 2018.3 same issue !

Tried also for VEC=8 and LANE=8: Basically same error code with local size adjusted
> Launching kernel lrn with local size: 1, 1, 12  (global size: 27, 27, 12)
> ERROR: ClEnqueueNDRangeKernel : CL_INVALID_WORK_GROUP_SIZE case 6

Mashahed, 2019-08-07:
@nslmike I am facing the same CL_INVALID_WORK_GROUP_SIZE issue. Did you find a solution?