doonny/PipeCNN #48: How to reduce RAMB utilization

HeroGian, 2018-02-19:
Dear doonny,
I'm trying to test PipeCNN framework on some Xilinx's FPGA embedded boards for take some power measurement. Actually I would like to compile the framework for Digilent ZedBoard but the synthetized program is too large for this FPGA platform, the XOCC compiler returns this error:

`297 RAMB18 and RAMB36/FIFO required but only 280`

Could you give me some hints for reducing the BRAM utilization? 

Thank You

doonny, 2018-02-22:
SDoC is not fully supported yet.
Use small value of VEC_SIZE and LANE_NUM can reduce the RAM costs.

HeroGian, 2018-02-26:
Thanks for the reply,

I have tried to reduce VEC_SIZE to 4 and LANE_NUM to 8 and it compiles perfectly!
But when I try to run the executable on ZedBoard it doesn't work at clEnqueueNDRangeKernel invocation for knl_lrn[i], the error is the following:

`Launching kernel lrn with local size: 1, 1, 24  (global size: 27, 27, 24)
ERROR: ClEnqueueNDRangeKernel : CL_INVALID_WORK_GROUP_SIZE case 6
ERROR: CL_INVALID_WORK_GROUP_SIZE 
Location: ../src/host/main.cpp:958
Failed to launch kernel lrn
WARNING: Profiling may contain incomplete information. Please ensure all OpenCL objects are released by your host code (e.g., clReleaseProgram()).
terminate called after throwing an instance of 'std::runtime_error'
  what():  software scheduler stopping while there are active commands
Aborted
`
probably I need to update some other parameters like LRN_MAX_LOCAL_SIZE or something related? could you give me some hints, I am not used OpenCL before

Thank You

doonny, 2018-02-28:
Sorry, we have not used SDSoC yet.

HeroGian, 2018-02-28:
Ok thank you very much anyway doonny.
I will keep trying to make it works with SDSoC, unfortunately I don't have a SDAccel license.

closed 2018-02-28 by HeroGian