doonny/PipeCNN #50: platform error

shawn2seo, 2018-02-22:
The following error was found in SDAcell version (SDK v2017.4) after executing make file under the project folder. 

* Error log ----------------------------------------------
ERROR: [XOCC 60-705] No device was found that matches 'xilinx:kcu1500:4ddr-xpr:4.0'. The supported devices are:
xilinx_vcu1525_dynamic_5_0
xilinx_kcu1500_dynamic_5_0

ERROR: [XOCC 60-587] Failed to add a device: specified platform xilinx:kcu1500:4ddr-xpr:4.0 is not found
Makefile:151: recipe for target 'conv.xclbin' failed
make: *** [conv.xclbin] Error 1
------------------------------------------------------------

I think the solution is to change 'xilinx:kcu1500:4ddr-xpr:4.0' to 'xilinx_vcu1525_dynamic_5_0'. 
But it is not sure that the PipeCNN's environment and behavior are still valid.

doonny, 2018-02-23:
Hi, again, 2017.4 is not tested yet. Please refer to the official user guide to due with the DSA version problem.  Better use v2017.2.

closed 2018-03-23 by doonny

sinatv52, 2019-07-23:
have you been able to compile with vcu1552 with newer versions of SDAccel?
I am going to test it with v2018.3 and aws-f1 FPGA.
but i am not familiar with the flow and makefileand etc 🤔