doonny/PipeCNN #52: Fixed-point model numbers

laski007, 2018-02-28:
Dear Prof. Wang, 
For the quantized parameter, you used (n,m) pair to denote the precision. For example, in VGG-16 1st layer, you used (8,7) (8, 0) (8,-2) to denote frac_w, frac_input and frac_output, for last FC layer, you used (8, 2) (8,2) (4,7). Is there any rules or constrains how to decide the numbers? Or just use any numbers you like? Can I change to other numbers? If I changed the fraction numbers and convert a new model, does it works or not? Thank you so much.

laski007, 2018-02-28:
Okay, I find
rule #1, previous layer's output fraction number must equal to next layer's input fraction number. 
rule #2, 1st layer's input fraction number is 0. 
Anything else? Why in AlexNet the 1st layer's output fraction is (8, -4) but in VGG16 the output is (8, -2)? For the weight fraction, sometime it is (8,7), sometime it is (8,8) or (4,7) or other numbers, why? Many thanks.

aazz44ss, 2018-03-01:
I also want to know, (N,m) denote the N bits which contain m fractional bits, and represent Nx2^-m.
Is N bits contain sign bits? 1st layer contain an integer between 0~255 which can be represent as (8,0) withou sign bit. what about the other layer?
and what is (4,7),(8,9),(8,10) mean? how can fractional bit larger than N word?

laski007, 2018-03-07:
Dear Prof. Wang,
Could you please help me to briefly describe how to decide the frac_w, frac_input and frac_output numbers when you have spare time? I'm still waiting ... Thank you so much. 跪谢Orz

aazz44ss, 2018-03-08:
frac_w, frac_input and frac_output depends on the distribution of your data.
If your weight is range from -32.0~32.0, like -19.83, 2.15, -5.14, 6.81, 21.02, 32.11.....
you should choose frac_w of 2.
there is 1 signed bit, 5 integer bits, 2 fractional bit.
you have to use 5 integer bits to contain your range of weight. ( 2^5 = 32 )
every number larger than 32 will become 32. (you can see conv.cl code)
and if your weight is range only from -1~1, you don't need so many integer bits,
so you can increase your fractional bit to increase precision.

fran_input and frac_output done by the same way.
you have to try fractional bit with different layers, and see the accurate with your test data set.

doonny, 2018-03-12:
Hi, all, please refer to matlab's fixed-point number's defination in the Fixed-point Toolbox. We are using a very similar data format.

closed 2018-04-01 by laski007