doonny/PipeCNN #53: Run time hang running on DE10 Nano

hammadj, 2018-03-01:
Hi there, I'm trying to get the basic version running on my DE10-nano board, however the execution hangs during Layer 1 execution and doesn't get further than the log below.

I'm using this for the linux image: https://github.com/thinkoco/c5soc_opencl

Any help would be much appreciated!
```
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

Platform: Intel(R) FPGA SDK for OpenCL(TM)
Using 1 device(s)
  Device 0: de10_nano_sharedonly_hdmi : Cyclone V SoC Development Kit
Device OpenCL Version: OpenCL 1.0 Intel(R) FPGA SDK for OpenCL(TM), Version 17.1
Device Max Compute Units: 1
Device Max WorkGroup Size: 2147483647
Device Max WorkItem Size: 2147483647
Device Global Memory Size: 512 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 1000 Mhz

Loading kernel/binary from file conv.aocx

61063552 total weights read 
1024 total output reference read 

154587 bytes image data read from binary files

Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel Pooling

Launching kernel MemWr with local size: 1, 1, 8  (global size: 27, 27, 96)

Launching kernel lrn with local size: 1, 1, 12  (global size: 27, 27, 12)
```

doonny, 2018-03-07:
@hammadj Hi, it seems that you are not using a BSP (de10_nano_sharedonly_hdmi ???) which is not officially supported. Please check if the official one works. Since, we have not met any problem on that.

thinkoco, 2018-03-10:
@hammadj @doonny Hi, all.  Sorry, it's my fault.These was something wrong with my SD card image. Now,this issue had be fixed.

closed 2018-03-12 by doonny