doonny/PipeCNN #57: ERROR: CL_INVALID_BINARY

ArjunaDeSoysa, 2018-03-14:
Dear Dear Prof. @doonny ,
I use De1-SoC and Angstrom v2014.12 - Kernel , I changed,
`const char *vendor_name = "altera";` in derictory ../project/host/main.cpp
because of when *vendor_name = "Intel" there was Error "Unable to find the desired OpenCL platform." 

now there is new error;
`Loading kernel/binary from file conv.aocx `
`ERROR: CL_INVALID_BINARY`

So i want to know, There are another things do i want to change ? (because of changing `const char *vendor_name = "altera";`)

Thank You

ArjunaDeSoysa, 2018-03-16:
Dear @thinkoco do you know how to slove this problem?

thinkoco, 2018-03-17:
@ArjunaDeSoysa  So, you have use the opencl sdk 16.0 runtime file and rebuild driver?
It's recommend to use sdk 16.1 runtime file.

ArjunaDeSoysa, 2018-03-17:
@thinkoco yes i use opencl sdk 16.1 runtime file in host computer. But in terasic site there's only have BSP(Board Support Package) for Altera SDK OpenCL 16.0. i saw that someone say that it is okay for this pipeCNN.

thinkoco, 2018-03-17:
@ArjunaDeSoysa   Also, you can use my SD card image [c5soc_opencl_lxde_all_in_one_180317.img](https://drive.google.com/open?id=1mAYHFvOw2xtgf-e8pntFCxCGOdaYNsgG).
Just need to replace the dtb with no vga one  [here](https://github.com/doonny/PipeCNN/files/1821204/socfpga_cyclone5_de1soc_novga.dtb.zip)
and delete the CL_CONTEXT_COMPILER_MODE_ALTERA=3  flag in the init_opencl16.1.sh file.

ArjunaDeSoysa, 2018-03-17:
Okay I'll try that. Thank you very much @thinkoco.

ArjunaDeSoysa, 2018-03-17:
@thinkoco , 1) Do i want to replace the socfpga_cyclone5_de1soc_novga.dtb with socfpga.dtb **or** de1soc_socfpga.dtb(inside the de1soc folder )?

2) I use UART to USB cable so , username : knat and password : knat right?

i am really sorry for the bothering again and again.

thinkoco, 2018-03-17:
@ArjunaDeSoysa  cover the socfpga.dtb with socfpga_cyclone5_de1soc_novga.dtb , and keep  the socfpga.dtb name. x2go user is knat,  root for uart.

ArjunaDeSoysa, 2018-03-18:
@thinkoco , Still i got same error,

```
root@C5SoC:~# source init_opencl_16.1.sh                                        
root@C5SoC:~# cd cnn                                                            
root@C5SoC:~/cnn# ./run.exe conv.aocx                                           
******************************************                                      
An OpenCL-Based FPGA Accelerator for CNNs                                       
******************************************                                      
                                                                                
Platform: Intel(R) FPGA SDK for OpenCL(TM)                                      
Using 1 device(s)                                                               
  Device 0: de1soc_sharedonly_vga : Cyclone V SoC Development Kit               
Device OpenCL Version: OpenCL 1.0 Intel(R) FPGA SDK for OpenCL(TM), Version 16.1
Device Max Compute Units: 1                                                     
Device Max WorkGroup Size: 2147483647                                           
Device Max WorkItem Size: 2147483647                                            
Device Global Memory Size: 512 MBytes                                           
Device Local Memory Size: 16 KBytes                                             
Device Max Clock Freq: 1000 Mhz                                                 
                                                                                
Loading kernel/binary from file conv.aocx                                       
ERROR: CL_INVALID_BINARY                                                        
Location: ../common/ocl_util.cpp:415                                            
Failed to create program with binary  
```
at the beginning(not only this time) there's not appear in this part below ;

```
61063552 total weights read 
154587 bytes image read 
1024 total output reference read 
```
i tried various thing. can you please help me to solve this?

ArjunaDeSoysa, 2018-03-18:
@thinkoco , you are right. i check `aocl board-xml-test` in the host computer, the result is in given below;
```
arjuna@arjuna-Inspiron-3542:~$ source /etc/profile
arjuna@arjuna-Inspiron-3542:~$  aocl board-xml-test 
 board-path       = /home/arjuna/intelFPGA/16.1/hld/board/terasic/de1soc

 board-version    = 13.0

 board-name       = de1soc

 board-default    = de1soc

 board-hw-path    = /home/arjuna/intelFPGA/16.1/hld/board/terasic/de1soc/./de1soc

 board-link-flags = -L/home/arjuna/intelFPGA/16.1/hld/board/terasic/de1soc/arm32/lib -L/home/arjuna/intelFPGA/16.1/hld/host/arm32/lib

 board-libs       = -lalterammdpcie -lstdc++

 board-util-bin   = /home/arjuna/intelFPGA/16.1/hld/board/terasic/de1soc/arm32/bin

```
The **bord-version is 13.0**. i think that is the problem. i download BSP(Board Support Package) in terrasic site. I serched BSP 16.1 to download and i coulden't find it.

Can you please send me the link , where can i download  BSP(Board Support Package) for Altera SDK OpenCL 16.1 .?

thinkoco, 2018-03-18:
@ArjunaDeSoysa  it seems to be  the board xml or hardware template version. It's not the SDK version. Before you building and running the host, check not setting the CL_CONTEXT_EMULATOR_DEVICE_ALTERA flag.Also ,some other flags.I have test the PipeCNN on my baord with my sd card image , It's OK. if  I set CL_CONTEXT_EMULATOR_DEVICE_ALTERA = 1 and run, get you same error. Maybe, not your same issue, but you can check it.

doonny, 2018-03-19:
you need firstly to reprogram the fpga manually on the DE1 platform

ArjunaDeSoysa, 2018-03-19:
Prof. @doonny did you mean,
run aocl program `aocl program /dev/acl0 conv.aocx` before `./run.exe conv.aocx` ?

doonny, 2018-03-19:
yes

doonny, 2018-03-19:
you could also use aocl diagnose to verify that you have the right driver installed.

ArjunaDeSoysa, 2018-03-19:
yes it worked. Thank you very much thinkoco and Prof. doonny

closed 2018-03-19 by ArjunaDeSoysa

Fnajjar, 2018-04-23:
Hi @doonny ,  @thinkoco , 
I am using the OpenCL SDK v17.1, and having the same error. 
What shall I do??

thinkoco, 2018-04-24:
@Fnajjar PCIE based board ?  you can try to test the runtime with the "aocl diagnose" or "aocl program" command first.

DasunBhanuka, 2018-04-24:
@thinkoco  @doonny  If I rename the folder 'de1soc_sharedonly' to 'de1soc', will the following get resolved?

```
Loading kernel/binary from file conv.aocx
ERROR: CL_INVALID_BINARY
Location: ../common/ocl_util.cpp:415
Failed to create program with binary
```

thinkoco, 2018-04-24:
@DasunBhanuka  No.
1. check the conv.aocx for which device by "aocl binedit conv.aocx get .acl.board board.txt && cat board.txt"
2. check the opencl.rbf for which device by "aocl diagnose"
they must be same device name.

Fnajjar, 2018-04-24:
Hey @thinkoco, @DasunBhanuka  
I tried the instruction but still have the same error
What do you think?

[image]

DasunBhanuka, 2018-04-24:
@thinkoco  @Fnajjar  @aazz44ss @doonny 

I am getting the exact same error just like this.. even though I am running **OpenCl 16.1**  Can you give us another idea?

thinkoco, 2018-04-24:
@Fnajjar  There may be some difference in 
1. aocl binedit conv.aocx list
2. "aocl binedit conv.aocx get .acl.board_spec.xml board.xml && cat board.xml"  or other file

Fnajjar, 2018-04-24:
@thinkoco , actually i did the commandes and it seems that there 's no ".acl.board_spec.xml " 

[image]

thinkoco, 2018-04-24:
@Fnajjar nothing in the .acl.board ?

Fnajjar, 2018-04-24:
@thinkoco, How shall I know? 
I tried "aocl binedit conv.aocx get .acl.board board.xml && cat board.xml" and I've got nothing :(

thinkoco, 2018-04-24:
@Fnajjar It seem to be an emulater .aocx file, there is no .acl.fpga.bin in the conv.aocx.
if emulater, try  "export CL_CONTEXT_EMULATOR_DEVICE_ALTERA=1" before running host

Fnajjar, 2018-04-24:
thank you @thinkoco , I did it and this time I have different error,
please find the image below! 
[image]

thinkoco, 2018-04-24:
@Fnajjar It's Ok,you need downlaod the  PipeCNN weight files [here](https://github.com/doonny/PipeCNN/tree/master/data).

Fnajjar, 2018-04-24:
@thinkoco, actually I have already uploaded them "/home/najjar/Documents/PipeCNN-master/data", *the problem is that it's not beeing detected or can't been found 
[image]

thinkoco, 2018-04-24:
@Fnajjar Also, data_alex is needed.see the host code and the path
```
const char *weight_file_path = "./data/data_alex/weights.dat";
const char *input_file_path = "./data/data_alex/image.dat";
const char *ref_file_path = "./data/data_alex/fc8.dat";
const char *dump_file_path = "./result_dump.txt";
```

DasunBhanuka, 2018-04-24:
Dear @thinkoco 

How to check this in fpga,

aocl binedit conv.aocx list
"aocl binedit conv.aocx get .acl.board_spec.xml board.xml && cat board.xml

as you said..

Fnajjar, 2018-04-24:
@thinkoco thank you very much for your help acctually it works! I already uploded the data_alex. 
I moved the files "conv.aocx" and the "run.exe" in the folloing path "/home/najjar/Documents/PipeCNN-master" and it worked! 
[image]
[image]

ArjunaDeSoysa, 2018-04-24:
@Fnajjar if you able to complete the HW emulation also you can help to @DasunBhanuka to get it work. cause you both have same problem right. Thank You.

DasunBhanuka, 2018-04-24:
@Fnajjar   Can you please explain  how you made it work, after that error..

Fnajjar, 2018-04-24:
@DasunBhanuka, well when I had the "ERROR: CL_INVALID_BINARY" I juste tiped the commande line "export CL_CONTEXT_EMULATOR_DEVICE_INTELFPGA" and the he still couldn't find the weights eventhough, I had the data in the wright directories.
So, I just moved the two files "conv.aocx" and the "run.exe" in the folloing path "/home/user/Documents/PipeCNN-master" and it worked!

[image]

DasunBhanuka, 2018-04-24:
@Fnajjar  Will you also try this using a FPGA hardware?

Fnajjar, 2018-04-24:
well now since it's working on emulator, I am going to implement it on a Attila Arria 10 Gx
but, I am trying to know how to do that?
Do you have an idea??

DasunBhanuka, 2018-04-24:
@Fnajjar  I am currently trying to implement it on Altera DE1SoC. If it works, I'll tell you..

thinkoco, 2018-04-24:
@DasunBhanuka Sorry for notice that  aocl diagnose on de1soc and aocl binedit  on PC

thinkoco, 2018-04-24:
@DasunBhanuka 
1.you can use "aocl diagnose" command after sourcing the opencl initial shell file on de1soc. 
2. then "aocl binedit conv.aocx list" and  "aocl binedit conv.aocx get .acl.board board.txt && cat board.txt" on PC
3. both the device names in step 1 and step 2 should be the same one.

DasunBhanuka, 2018-04-24:
@thinkoco  I did as you said,
[image]
It is not same..

DasunBhanuka, 2018-04-24:
Dear @thinkoco 
I did as you said..
[image]
Here's the result

DasunBhanuka, 2018-04-24:
Dear @thinkoco why do you think I get the error `No such file or directory` Or Am I missing something here?

thinkoco, 2018-04-25:
@DasunBhanuka Hi,  you can cd to the folder which contians conv.aocx first .Also,read permissions of this file

DasunBhanuka, 2018-04-26:
@thinkoco Can you send me the files you created for DE1SoC? conv.aocx & run.exe
Will they be compatible with my board?

Frankzd, 2019-03-19:
Hi@Fnajjar，I am using Arria 10 Gx, and 17.1 SDK just like you. And I am facing the problem you might met before 

[image]

Do you have any idea how to solve this problem?

Frankzd, 2019-03-19:
Hi @Fnajjar，I am using Arria 10 Gx, and 17.1 SDK just like you. And I am facing the problem you might met before. Do you have any idea how to solve this problem?