doonny/PipeCNN #63: ERROR: CL_INVALID_BINARY

Meyerjan, 2018-03-26:
Dear Prof. @doonny  ,
there is an error "ERROR: CL_INVALID_BINARY",when i run the project.
i saw the problem like this,but i run aocl program /dev/acl0 conv.aocx before ./run.exe conv.aocx
Do you some ideas about that.
thank you.  

root@socfpga:~# source init_opencl_16.1.sh                                        
root@socfpga:~# aocl program /dev/acl0 conv.aocx 
aocl program:Running reprogram from /home/root/opencl_arm32_rte/board/c5soc/arn
Reprogramming was successfull                                                           
root@socfpga:~# ./run.exe conv.aocx 
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs
***************************************************
Platform: Altera SDK for OpenCL                                                           
Using 1 device(s)                                                                         
  Device 0: de1soc_sharedonlyCyclone V SoC Development Kit                                
Device OpenCL Version: OpenCL 1.0 Altera SDK for OpenCL, Version 16.0                     
Device Max Compute Units: 1                                                               
Device Max WorkGroup Size: 2147483647                                                     
Device Max WorkItem Size: 2147483647                                                      
Device Global Memory Size: 512 MBytes                                                     
Device Local Memory Size: 16 KBytes                                                       
Device Max Clock Freq: 1000 Mhz 

Loading kernel/binary from file conv.aocx 
ERROR: CL_INVALID_BINARY
Location: ../common/ocl_util.cpp:415
Failed to create program with binary

ArjunaDeSoysa, 2018-03-26:
@Meyerjan use the BSP 16.1 and try it again. all information are include in issue #57 by thinkoco.

Meyerjan, 2018-03-26:
thanks for you,but i also can't find sdk 16.1.

Meyerjan, 2018-03-26:
@ArjunaDeSoysa and you mean it is right 
"you can use my SD card image c5soc_opencl_lxde_all_in_one_180317.img.
Just need to replace the dtb with no vga one here
and delete the CL_CONTEXT_COMPILER_MODE_ALTERA=3 flag in the init_opencl16.1.sh file."
thank you so mach

ArjunaDeSoysa, 2018-03-26:
@Meyerjan you can download by simply  click that link he gave. then you want to replace socfpga.dtb  from socfpga_cyclone5_de1soc_novga.dtb and rename it as socfpga.dtb.

Meyerjan, 2018-03-26:
thank you so much,i will try it

Meyerjan, 2018-04-01:
@ArjunaDeSoysa 
hi, I try,but i got same error,what do you do,when you got same error.
I'm so sorry, I'm always in trouble with you.

ArjunaDeSoysa, 2018-04-01:
@Meyerjan  you can check your env variable again and if you are sure those are correct. you can put your email address. so i can send code which i used for one part.

Meyerjan, 2018-04-01:
@ArjunaDeSoysa
 my email is mjdyx2014@163.com
thank you

closed 2018-04-14 by Meyerjan