doonny/PipeCNN #64: Fixed point quantization input/output data

hammadj, 2018-03-31:
I'm working on implementing YOLO with PipeCNN, and was wondering how to determine the correct precision_config for these layers. For the weights, it is easy to determine the range of values from the caffemodel file, but for `frac_din` and `frac_dout` I am not sure how to determine these values. I read the quantization flow in the paper, but don't know how I should actually perform those steps. Any help would be much appreciated.

doonny, 2018-04-11:
The featuremap should be truncated at runtime. You need modification on Darknet itself.
I don't think either Keras nor TensorFlow support fixed-point convolutions, you really need to write your own code for fixed-point computations. We have done this in Matlab, writing everything from scratch using fixed-point toolbox.

doonny, 2019-03-11:
I am assuming this problem has been solved.

closed 2019-03-11 by doonny