doonny/PipeCNN #69: Some problems encounted when compiling using Deo-Nano-SoC board

roddregs, 2018-04-16:
@doonny 
Hi Sir, good day!
Tried to compile PipeCNN in Deo-Nano-SoC board.
However, I encountered the following problem when running the main makefile:
g++   ./host/main.o ../common/ocl_util.o ../common/timer.o -o run.exe -L/thesis/intelFPGA/16.1/hld/board/de1soc/arm32/lib -L/thesis/intelFPGA/16.1/hld/host/arm32/lib -L/thesis/intelFPGA/16.1/hld/host/linux64/lib -Wl,--no-as-needed -lalteracl -lalterahalmmd -lalterammdpcie -lelf 
/usr/bin/ld: cannot find -lalterammdpcie
collect2: error: ld returned 1 exit status
Makefile:118: recipe for target 'run.exe' failed
make: *** [run.exe] Error 1
Is Deo-Nano-SoC board compatible with the configurations of PipeCNN?
Thank you very much in advance...

doonny, 2018-04-18:
Do you mean DE10-nano ?

roddregs, 2018-04-18:
Sir @doonny, it's De0-nano board, cyclone 5 from terasic.

doonny, 2018-04-19:
I don't think there is a OpenCL BSP for DE0-nano board.

zhenyu707, 2018-04-20:
Hello @doonny  @roddregs ,

I am trying to run this in De10-nano but there is some error. May I know how to solve this? 
[image]

For your information, threre is a OpenCL BSP for De10-nano and this is the link : 
[http://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&No=1046&PartNo=4](url)

doonny, 2018-04-20:
The cyclone device does not support emulator model.

zhenyu707, 2018-04-20:
Hello @doonny ,

Would this error be solved if I run on De10-Standard?

Thank you.

doonny, 2018-04-22:
All SoC-FPGA does not support emualtor model. You need to use PCIE-based bsp.

CrazyBingo, 2018-05-11:
# @
[image]
the same as you,  @zhenyu707 , have you ever resolve it?(my board is de10 .too)

zhenyu707, 2018-05-11:
@CrazyBingo  Hello, the problem still can not be solved.

closed 2019-03-11 by doonny