doonny/PipeCNN #73: Implement 3-dimensional (3-D) multiply-accumulate (MAC) by using a HLS-friendly 1-D convolution structure which flattened the 3-D convolution

ArjunaDeSoysa, 2018-04-24:
Dear @aazz44ss, @doonny  I have a problem in the topic as i said, how can convert in 3-D  MAC using 1D and the what is the procedure in multiplier adder tree. Can you please tell me the whats the procedure of these. If you are busy can you please send me the some links or resource regarding above things.

Thank You

ArjunaDeSoysa, 2018-04-25:
Dear @doonny , @aazz44ss , @hammadj can i know the what is these parameters are? Please help me to find outthis one. Thank you.
[image]

aazz44ss, 2018-05-02:
maybe you need to read some document about convolution .

ArjunaDeSoysa, 2018-05-04:
Okay i got it. Thank you.

closed 2018-05-04 by ArjunaDeSoysa