doonny/PipeCNN #74: how to run the exemple on my card ???

Fnajjar, 2018-04-27:
I want to test the Pipe CNN on an Attila Arria 10 GX card with an Intel SDK open CL v17.1. I did the implentation and it's working!
Stil, I want to do the demo on my card:
hwo can I do that?
 - Shall I download the imagNet database? how can I do that 
 - Shall I upload caffe? and since I am using ubunto 16.04 shall I upload cuda toolbox 8.0?
Please help! I wan't to simulate the demo!

thank you very much for the further explanation.

manojrohit, 2018-05-04:
Hello @Fnajjar 

Install the Intel SDK for openCL. This guide might help you.(https://www.altera.com/en_US/pdfs/literature/hb/opencl-sdk/aocl_getting_started.pdf)

Install OpenCV.

As this repo supports Intel opencl v16.1, make necessary changes for v17.1. 

Download the imgaenet 2012 validation set.

Fnajjar, 2018-05-04:
Thank you very much for your answer @manojrohit, but how can I download the imgaenet 2012 validation set??
I didn't find it in the ImageNet website unfortunatly

manojrohit, 2018-05-04:
You can download by registering in http://www.image-net.org/download-images.

Fnajjar, 2018-05-07:
Thank you @manojrohit,  
So ones I upload the validation images 2012 from the imageNet, do I extract the folder in the PipeCNN-master/data/imagenet?
Is there modifications to make in the file's code?

manojrohit, 2018-05-07:
If you go through the host cpp file, in line 71 you can see 
char  picture_file_path_head[100] ="/home/dwang/Work/imagenet/ilsvrc2012/ILSVRC2012_img_val/ILSVRC2012_val_";

You can extract the validation data and change the path in host program.

Fnajjar, 2018-05-09:
@manojrohit thank you so much it's finally working 
that was so helpful :)

doonny, 2018-05-10:
Definitely YES : )

Fnajjar, 2018-05-11:
Thank you @doonny  for this great news! :)
Do you have previous work or some advices on how shall I proceed? 
Thank you very much for the help

roddregs, 2018-06-08:
@Fnajjar @manojrohit @doonny 
I was using DE1-SoC Cyclone V board and successfully compiled and run in the board the PipeCNN using PLATFORM=arm32 and flow=hw which only read one image file.
And now, I tried to enable OpenCV to experiment loading multiple images from Imagenet 2012 validation set.
I set USE_OPENCV = 1 in the Makefile and installed OpenCV 2.4.9 in my PC (Ubuntu 16.04).
When I compiled, I got the following error that says:
 libopencv_core.so :file not recognized: File format not recognized

[image]
Any suggestion on how to solve this error? Is OpenCV 2.4.9 not compatible with the current set-up of PipeCNN?
Thank you very much.

doonny, 2018-06-13:
First, you need to cross-compile your opencv lib for arm. Your opencv installed is for PC not for arm

arun0303, 2018-08-02:
Hi Sir,
how to change cross compile OpenCV  lib for arm?

arun0303, 2018-08-02:
Hi Sir,
how to cross compile OpenCV lib for arm?

Wanghaoshu, 2018-09-09:
How to use VNC for image visualization on de10_nano board @doonny @manojrohit @roddregs

doonny, 2019-03-11:
I am assuming this problem has been solved.

closed 2019-03-11 by doonny