doonny/PipeCNN #77: ERROR:  usr/local/lib/libopencv_co.so: file not recognized: file format not recognized

DasunBhanuka, 2018-05-03:


DasunBhanuka, 2018-05-03:
@doonny @thinkoco @aazz44ss @hammadj, which OpenCV are you using?

DasunBhanuka, 2018-05-04:
I am sorry, I meant OpenCV, I've tranferred from OpenCV 4.0.0 to OpenCV 3.3.1. But still gets the same error..  `arm-linux-gnueabihf-g++ .. usr/local/lib/libopencv_co.so: file not recognized: file format not recognized`. After a while, I found that this is due to an issue in building OpenCV for ARM..

DasunBhanuka, 2018-05-04:
Dear  @doonny,
This is the error I get. Can you tell me **which OpenCV version you used** for that, so that I can try it also and see just in case..
[image]

roddregs, 2018-06-10:
@DasunBhanuka 
Were you able to solve this error? Encountered same error here..
How do you solve it?
Thanks.

Enshall, 2018-07-04:
file format issues always mean ur compiler was wrong.
ur path of opencv in makefile  is /usr/local/opencv-arm,  but what actually works is  usr/local/lib/..
so, the answer is just in ur logs.

DasunBhanuka, 2018-07-09:
@Enshall  Were you able to successfully implement the PipeCNN on a DE1SoC including OpenCV part?

Enshall, 2018-07-09:
@DasunBhanuka Yeah, I made it.

DasunBhanuka, 2018-07-09:
@Enshall  Can you please tell me what you did to solve this issue ? I am stuck with this for weeks.. It would be a great help..

DasunBhanuka, 2018-07-09:
@Enshall Did you encounter the same issue with DE1SOC.. ?  In case you want, my email is lokubaba321@gmail.com

Enshall, 2018-07-09:
@DasunBhanuka 
I don`t have a problem with u or opencv, but I guess ur opencv compilation didn`t success(/usr/local/opencv-arm exits?).
what now actually works is ur x86 version, arm-gcc can`t recognized.
so recompile ur opencv with arm-gcc or just download a copy may be worked.

DasunBhanuka, 2018-07-09:
@Enshall 
Yes, Problem might be in "arm-gcc" . Couldn't find a compatible version for DE1SOC..
How did you install arm-gcc ?

SudharanTJ, 2018-07-10:
@Enshall 
I am having the same problem as well. Did you get the same error as well.. ? The problem is in the `arm-linux-gnueabihf` How did you get it to work ?

Enshall, 2018-07-13:
@DasunBhanuka
ur compiler is ok. 
ur log says there is no opencv_core.lib in /usr/local/opencv-arm,  that`s means u didn`t corss-compile opencv successfully or ur path in makefile is wrong. u can check them first.  
@SudharanTJ
Cross compile opencv with CMake, modify opencv path in ur makefile (OCV_INCLUDES OCV_LIBDIRS).

arun0303, 2018-08-01:
hi,
I am using DE10 Standard kit i am facing the following error kindly help me. 

Thanks 
ARUN
[image]

arun0303, 2018-08-01:
hi @DasunBhanuka  kindly help me on this

doonny, 2019-03-11:
I am assuming this problem has been solved.

closed 2019-03-11 by doonny