doonny/PipeCNN #78: Cannot Launch lrn kernel in Layer 0

manojrohit, 2018-05-04:
Hello @doonny 

I am obtaining the following error when I run this on Arria10 :

Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel Pooling

Launching kernel MemWr with local size: 1, 1, 64  (global size: 27, 27, 128)

Launching kernel lrn with local size: 1, 1, 6  (global size: 27, 27, 6)
ERROR: CL_INVALID_EVENT 
Location: host/main.cpp:870
**Failed to finish lrn event**

My VEC_SIZE is 16, LANE_NUM is 64, CONV_GP_SIZE_X is 7. My board model is   
https://www.altera.com/products/boards_and_kits/dev-kits/altera/kit-a10-gx-fpga.html

Do you think I have used many resources?

manojrohit, 2018-05-04:
I think this issue occurs multiple devices. With single device I face no issue. :)

Enshall, 2018-06-11:
Hi manojrohit doonny @manojrohit  @doonny
I met a CL_INVALID_WORK_GROUP_SIZE error when Launching kernel MemWr.
Can you give me some advice ?
Thank you all the time !

Executing Layer 1:
Launching single work-item kernel winbuffer
Launching single work-item kernel Conv
Launching single work-item kernel Pooling
Launching kernel MemWr with local size: 1, 1, 2  (global size: 27, 27, 96)
ERROR: CL_INVALID_WORK_GROUP_SIZE
Location: host/main.cpp:846
Failed to launch kernel memWr

Enshall, 2018-06-11:
Hi @doonny @manojrohit, I get a 'CL_INVALID_WORK_GROUP_SIZE' error when clEnqueueNDRangeKernel.
my log is below,
Launching kernel MemWr with local size: 1, 1, 2  (global size: 27, 27, 96)
ERROR: CL_INVALID_WORK_GROUP_SIZE
Location: host/main.cpp:846
Failed to launch kernel memWr
I`m running with opencv.
Hope for you advices. Thank you very much!

closed 2018-08-20 by doonny