doonny/PipeCNN #86: Can't fit design on DE1-SoC Board

KamelAbdelouahab, 2018-06-21:
Thanks for the great work guys ! 
I'm trying to reproduce your results on a DE1-SoC device. However, it seems Quartus can't fit the the design on the board. 
``C:\Users\Kamel\dev\PipeCNN\project> aoc -v -report -board=de1soc_sharedonly -I device/RTL -L device/RTL -l rtl_lib.aoclib ./device/conv_pipe.cl -o bin/pipe_cnn.aocx``
``aoc: Selected target board de1soc_sharedonly``
``aoc: Running OpenCL parser....``
``aoc: OpenCL parser completed successfully.``
``aoc: Optimizing and doing static analysis of code...``
``aoc: Linking with IP library ...``
``Checking if memory usage is larger than 100%``

``!===========================================================================``
``! The report below may be inaccurate. A more comprehensive``
``! resource usage report can be found at conv_pipe/reports/report.html``
``!==================================================================``=========``

``+--------------------------------------------------------------------+``
``; Estimated Resource Usage Summary                                   ;``
``+----------------------------------------+---------------------------+``
``; Resource                               + Usage                     ;``
``+----------------------------------------+---------------------------+``
``; Logic utilization                      ;  100%                     `;``
``; ALUTs                                  ;   64%                     ;``
``; Dedicated logic registers              ;   42%                     ;``
``; Memory blocks                          ;  110%                     ;``
``; DSP blocks                             ;   31%                     ;``
``+----------------------------------------+---------------------------;``
``aoc: First stage compilation completed successfully.``
``Compiling for FPGA. This process may take a long time, please be patient.``
``Error (170012): Fitter requires 4594 LABs to implement the design, but the device contains only 3207 LABs``
``Error: Cannot fit kernel(s) on device``
Anyone of you tried with the DE1-SoC board ? or is it related to Quartus 18?

manojrohit, 2018-06-21:
Try decreasing the LANE_NUM in project/device/hw_param.cl

KamelAbdelouahab, 2018-06-22:
Thanks for the help, I was able to compile the Kernel for DE1-SoC with LANE_NUM =  4
Could tweak this value to have better throughput I guess. I will investigate that 👍

closed 2018-06-22 by KamelAbdelouahab

SudharanTJ, 2018-07-09:
@doonny @manojrohit  @KamelAbdelouahab 
Thanks for the help.. But now I'm trying to figure out how to implement Kernal including OpenCV part on DE1SoC.. Getting errors..