doonny/PipeCNN #88: how to generate the fc8 file

Fnajjar, 2018-07-03:
hi, 
I am trying to run an other image database with the Pipecnn.
I generated the Weights.dat but i am having wrong recognitions, and probabilities.
So I suspected a problem with the fc8 file.
Could you please tell me, how to generate an fc8 file for the new dataset?

manojrohit, 2018-07-04:
I think the procedure is mentioned in https://github.com/doonny/PipeCNN/tree/master/data

manojrohit, 2018-07-06:
You are on OpenCV mode. fc8.dat is output from the inference of image.dat, which I bielieve is the binary of (https://github.com/doonny/PipeCNN/tree/master/data/picture). 

Are you using the same  https://github.com/doonny/PipeCNN/blob/master/data/imagenet/synset_words.txt?

Fnajjar, 2018-07-06:
hey @manojrohit , thank you for the answer
No actually, since I am using the german traffic sign recognetion database GTSRB, I am trying to adopt the PipeCNN to treat it I uploaded a caffemodel I converted the weights thanks to the matlab script, then used the labels for the new base, val.txt too

but still having wrong recognition, no probabilities nor accuracy.
besides, if I put the size of the new weights file it does not work, only with the size of imagenet weight's size it runs

[image]
[image]
[image]
[image]
[image]

I think there are parameters to change but I don't know!
any ideas please

Fnajjar, 2018-07-09:
hey @manojrohit , thank you for the answer
No actually, since I am using "the german traffic sign recognetion database GTSRB", I am trying to adopt the PipeCNN to treat it I uploaded a caffemodel I converted the weights thanks to the matlab script, then used the labels for the new base, val.txt too

but still having wrong recognition, no probabilities nor accuracy.
besides, if I put the size of the new weights file it does not work, only with the size of imagenet weight's size it runs

[image]
[image]
[image]
[image]
[image]

I think there are parameters to change but I don't know!
any ideas please

manojrohit, 2018-07-10:
I think there is a small change from original caffe net. The MATLAB script would generate weights according to fc8-1000 because original caffenet output layer has 1000 and not 1024. This might be the problem.

doonny, 2018-07-10:
@manojrohit thx for the answer.

Fnajjar, 2018-07-11:
Hi @manojrohit and @doonny thank you very much for the answers.
I replaced 1024 with 1000 in the layer_config.h, and in the main.cpp; and choose the weights according to fc8-10000, but no changes acutally I am still having wrong answers(all inferences are false , wrong probabilities and false labels), and no accuracy 
here below you find some screen shot first for the 1024 config
[image]
[image]
[image]
[image]

and these are for the 1000 config 

[image]
[image]
[image]
[image]

Thank for your exchange

manojrohit, 2018-07-11:
The errors might be because of the way you have trained the model. Did you use CaffeNet and not AlexNet? Some layers are different in both.

My suggestion is to perform a CPU emulation for one particular image in the database and compare the outputs layerwise with FPGA.

Fnajjar, 2018-07-11:
Well actually I used a pre trained model GTSRB-caffe-model (https://github.com/magnusja/GTSRB-caffe-model) using the deep learning model AlexNet. 
Meantime, I am going to try your suggestion.
Thank you so much @manojrohit

manojrohit, 2018-07-11:
PipeCNN is performing the inference of caffenet. The difference between caffenet and alexnet is the order of pooling and norm layers in Layer 1 and Layer 2.

Fnajjar, 2018-07-11:
@manojrohit first of all I want to thank you for your answers, you were right.
I checked the deploy.prototxt and the order of the 2 first layers was reversed. That makes sense, and it's much more clear.
Otherwise, I was wondering if there are some changes to do in pipeCNN code to adapt it to the Alexnet model, since in the documentation it's always written (AlexNet(CaffeNet)) seems like they are similar however they are  not.
 
Or, what can you advise me? shall I retrain the model with caffenet??

@manojrohit thank you so much again for the valuable help :)

[image]

manojrohit, 2018-07-12:
I think you should retrain the model to CaffeNet. It would take some effort to change it to AlexNet.

Fnajjar, 2018-07-19:
Hey all; 
I am new to caffe and I am training the model with caffenet, and I was wondering If it's okay that I am having this kind of results
Thank you for your advices!! It's so helpful :)
[image]
[image]

doonny, 2019-03-11:
Hi, please refer to the README in the data folder.

closed 2019-05-08 by doonny