doonny/PipeCNN #89: Adding 1 and shifting the last bit?

manojrohit, 2018-07-06:
Hello @doonny 

Why are we not right shifting (frac_w+frac_din-frac_dout) directly? Why are we right shifting (frac_w+frac_din-frac_dout -1) and right shifting 1 at the end?

Current Version :- 
conv_with_rnd_bit[ll] = (conv_sign_exten[ll] | (conv_out[ll]>>(frac_w+frac_din-frac_dout**-1**))) + **0x01**;

if(conv_with_rnd_bit[ll]>=256)
	conv_sum_bias[ll] = MASK9B & 0xFF;
else if(conv_with_rnd_bit[ll]<-256)
	conv_sum_bias[ll] = MASK9B & 0x100;
else
	conv_sum_bias[ll] = (MASK9B & conv_with_rnd_bit[ll])+(bias[ll]>>(frac_w-frac_dout-1))+0x01;

conv_final[ll] = MASK8B & (conv_sum_bias[ll]>>**0x01**);

Why is not?
conv_with_rnd_bit[ll] = (conv_sign_exten[ll] | (conv_out[ll]>>(frac_w+frac_din-frac_dout)));

if(conv_with_rnd_bit[ll]>=128)
	conv_sum_bias[ll] = MASK8B & 0x7F;
else if(conv_with_rnd_bit[ll]<-128)
	conv_sum_bias[ll] = MASK8B & 0xFF;
else
	conv_sum_bias[ll] = (MASK9B & conv_with_rnd_bit[ll])+(bias[ll]>>(frac_w-frac_dout))

aazz44ss, 2018-07-06:
in order to have rounding

manojrohit, 2018-07-06:
We are adding 0x01 twice. 

After accumulating the weights : //I think this is not required 
conv_with_rnd_bit[ll] = (conv_sign_exten[ll] | (conv_out[ll]>>(frac_w+frac_din-frac_dout -1 ))) + 0x01;

After adding the bias : 
conv_sum_bias[ll] = (MASK9B & conv_with_rnd_bit[ll])+(bias[ll]>>(frac_w-frac_dout-1))+0x01;

For rounding I think its enough to add 0x01 once. It would be effective if we just add it after the bias.

doonny, 2018-07-10:
I have introduced rounding twice to keep the same precision with our HW-emulation script. You can modify this part for only one rouding.

closed 2018-08-20 by doonny