doonny/PipeCNN #94: LRN piece-wise linear approximation function coefficients

Mario1112, 2018-08-20:
Hello,

I read your paper and also studied the code [here [1]](https://github.com/doonny/PipeCNN/blob/e63a9b7a7dc55e651193a85c474086f3e08f0835/project/device/conv_pipe.cl#L772) and [here [2]](https://github.com/doonny/PipeCNN/blob/e63a9b7a7dc55e651193a85c474086f3e08f0835/project/device/hw_param.cl#L80-L83) but I am wondering how the values of `h_inv`, `coef0` and `coef1` are calculated. From the paper I can understand `x_sample` is derived from the `2^(-n)` value with `n = 2`, but using only this subdivision I can at best get an maximum error of 8.432119111524738% as opposed to your value of a maximum error of 0.5%. Can you elaborate on how the values for `h_inv`, `coef0` and `coef1` are computed? Thank you.

[1] https://github.com/doonny/PipeCNN/blob/e63a9b7a7dc55e651193a85c474086f3e08f0835/project/device/conv_pipe.cl#L772
[2] https://github.com/doonny/PipeCNN/blob/e63a9b7a7dc55e651193a85c474086f3e08f0835/project/device/hw_param.cl#L80-L83

doonny, 2018-08-20:
Hi, pls refer to the paper for more detailed informations.

Mario1112, 2018-08-20:
Thank your for your fast reply. Unfortunately I cannot find this information in the paper. In section 2.B. part 4) "LRN Kernel" you describe 

> we divide the function evaluation range by using power of 2^-n, where n is an integer that controls the accuracy.

and

> In AlexNet implementation, a maximum approximation error of 0.5% is achieved by setting n = 2.

Can you help me understand in which section the information is given? I could not find it in the paper. Thank you for your time and effort.

closed 2018-10-23 by Mario1112