doonny/PipeCNN #95: DE1-SoC parameters

0x47, 2018-10-23:
Hello,

In the [README.md](https://github.com/doonny/PipeCNN/blob/master/README.md#L34) file it says that the execution time for AlexNet on the DE1-SoC is 150 ms. Can you provide the configuration for the hw_param.cl file to achieve 150 ms? I tried with `VEC_SIZE = 4`, `LANE_NUM = 4`, `CONV_GP_SIZE_X = 7` and `PIPE_DEPTH = 6` but I can only achieve 351.607 ms. Increasing `LANE_NUM` further (e.g. 16 as in the default setting) is not possible because of the low hardware resources of the DE1-SoC.

aazz44ss, 2018-10-23:
It also depend on Fmax and batch size you use.
What is your Fmax?

0x47, 2018-10-23:
The batch size is 1 and the kernel clock frequency is `139.56 MHz` taken from the `quartus_sh_compile.log` file.

0x47, 2018-10-24:
It turns out that removing the `--profile` parameter from the Makefile solves this issue and allows setting `VEC_SIZE = 8` and `LANE_NUM = 8`. Thanks and have a nice day!

closed 2018-10-24 by 0x47