doonny/PipeCNN #96: Power Consumption on DE1-SoC

0x47, 2018-10-25:
Hello,

In you paper you are measuring the performance of PipeCNN using the DE1-SoC device (among others) and it shows a power consumption of 2.1 W in table 1 (page 282). I am trying to replicate this evaluation but my measurement results between 5.9 W and 6.7 W using the official 16.0 BSP. Even the idle board uses approximately 5.7 W (just after booting). During this test only USB-UART is connected to show the Linux terminal output.

Could you kindly let me know how you achieved this low power consumption using the DE1-SoC board? Did you change any BSP or Linux configuration?

Thank you very much.

doonny, 2018-10-26:
Here you are
[image]

0x47, 2018-10-27:
Thank you for your quick reply. Can you confirm that you booted the Terasic 16.0 BSP to run the PipeCNN run.exe file in the Ångström Linux on the SD-card?

Mario1112, 2018-11-02:
[This forum post](https://forums.intel.com/s/question/0D50P00003yyRc5SAE/de1soc-power-consumption?language=en_US) suggests the DE1-SoC running the Linux BSP with Ethernet cable (as in the first picture) draws 0.51 A * 12 V = 6.12 W.

> fpga configured from HPS with SD with Linux Console with ethernet cable connected 0.51A 

Even with only the FPGA running it uses 3.6 W

> With fpga only are 12*0.3=3.6W, a lot of power. 

I hope we can get a clarification because running at 2.1 W would be much better.

doonny, 2018-11-19:
Hello, we are using BSP 16.0 from Terasic and also the linux kernel provided with the BSP. And make sure the arm processor is not over clocked.

A power meter like the following one shoud give you the correct result:
[image]

And a quick search will show that 2.0~3.0W is the ordinary power consumption for cyclone-V device
[image]

https://support.criticallink.com/redmine/projects/mityarm-5cs/wiki/Power_Supply_Requirements

I hope the information helps.

doonny, 2018-11-19:
Also by my knowledge, the usb and jtag are also power consuming. better do the measurement without all the peripherals. Good Luck.

doonny, 2019-03-11:
I am assuming this problem has been solved.

closed 2019-03-11 by doonny