doonny/PipeCNN #97: difference between opencl for gpu and opencl for fpga

Thilanka97, 2018-11-14:
@doonny  do you have an idea about the difference between, coding targeting fpga using opencl and coding targeting GPU and CPU using opencl ? How different are they ? what are the main differences. If you could help me with this, it would be a great help. please be kind enough to help me.

Thanks in advance!

doonny, 2018-11-14:
OpenCL for FPGA is a new way for modeling macro architecture/digital circuit in high-level programming langurage
OpenCL for GPU is still parallel programing on SIMD processor

Thilanka97, 2018-11-15:
@doonny Thank you very much for the reply! I am trying to implement yolo or tiny yolo on fpga using opencl, for that I am gonna use a yolo opencl implementation that was coded to run on gpu and cpu. But I dont have much of a idea about what changes do I need to make to be able to run on an fpga. 

I am trying to read and understand your code also so that I will be able to get an idea of how to implement cnn on opencl. Do you have an idea about this ? 

Thanks in advance!

doonny, 2018-11-19:
Hi, you will find a proper way to implement YOLO based-on PipeCNN for sure, because we have verified that. The first thing to do is to fully understand the computation flow of YOLO and PipeCNN,.

Thilanka97, 2018-12-27:
@doonny thanks for your valuable advice! Now I am trying to understand the flow of yolo and pipecnn. from what I understand so far, I need to add more kernel functions to the kernel code other than the main 4 kernel functions you have for alexnet and vgg net. And also I need to change layer config file. And have to change the host file according to the changes I make in the kernel code right ? Am I correct ? Is there anything other than this that I need to do ? with these changes would I be able to run tiny yolo real time of fpga ? Please guide me. 

Thanks in advance!

doonny, 2019-03-11:
Basicly for YOLO, you only need to modify the convolution kernel to accelerator the algorithm, since the conv operations consumes most of the execution time. You don't really need new kernels.