doonny/PipeCNN #98: Performance On KCU1500 with SDx Environment

yuanm2, 2018-11-29:
We successfully execute the PipeCNN's Alexnet model and here is our performance summary for testing first 10 images from imagenet. We notice that the coreConv kernel takes 339.24ms on average and macPool 521.841ms. Does anyone have the similar results?

[performance_sdx.pdf](https://github.com/doonny/PipeCNN/files/2630314/performance_sdx.pdf)

doonny, 2019-03-11:
Anyone has optimization suggestions for Xilinx FPGAs ?

cococener, 2019-12-13:
orry,I have a question,I want to know if this project can be compiled by SDAccel ,Because when I compiled it ,it can't produce .xclbin,I used HW and System mode,just don't have the .xclbin,and also build finished.

cococener, 2019-12-13:
My

> We successfully execute the PipeCNN's Alexnet model and here is our performance summary for testing first 10 images from imagenet. We notice that the coreConv kernel takes 339.24ms on average and [...]
Sorry,I have a question,I want to know if this project can be compiled by SDAccel ,Because when I compiled it ,it can't produce .xclbin,I used HW and System mode,just don't have the .xclbin,and also build finished.And if you can ,can you tell me what change do you do for the pipe project,Because my board is KCU1500,too.

closed 2020-12-17 by doonny