doonny/PipeCNN #113: Failed to create bn kernel on Resnet

WenyeLiu, 2019-08-11:
Hi, I have test pipecnn on the Terasic OpenVINO Starter Kit (OSK) which can run the Alexnet smoothly. While when I change the pipecnn for RESNET in hw_param.cl file, it will report the "failed to create bn kernel" message. I only change the network definition to RESNET in hw_param.cl and recompile the project. Any extra setting required for RESNET?

doonny, 2019-09-19:
Please check the resource utilization in the report. There might not be enough hardware resource for the cyclone device

WenyeLiu, 2019-09-21:
Thanks. I change the card to another PC and tune some hw params. Now it can run.

closed 2019-09-21 by WenyeLiu

mahinlma, 2019-12-02:
@WenyeLiu Facing same issue. Could you please share, what changes you made?

WenyeLiu, 2019-12-03:
You can change the VEC_SIZE and LANE_NUM to a smaller value