doonny/PipeCNN #114: Unable to compile with 19.2 version

ignvinay, 2019-09-09:
Hi doony,
     I am trying to compile PipeCNN with opencl sdk for intel FPGA 19.2 version.
I am finding following errors :
One small change in Makefile i did was : emulator changed to emulator -legacy-emulator.
I am getting error in opencl file for macro with Item:

Can you please help me resolve this.

vinay@ign:~/Documents/opencl_fpga/PipeCNN/project$ make
aoc -v -report -march=emulator -legacy-emulator -I device/RTL -L device/RTL -l rtl_lib.aoclib  -g device/conv_pipe.cl -o conv.aocx
Resolving library filenames to full paths
  lib_path = device/RTL
    lib_file = rtl_lib.aoclib
Resolved rtl_lib.aoclib to /home/vinay/Documents/opencl_fpga/PipeCNN/project/device/RTL/rtl_lib.aoclib
  lib_path = .
All OpenCL C models were extracted from specified libraries and added to compilation
aoc: Environment checks are completed successfully.
aoc: Cached files in /var/tmp/aocl/ may be used to reduce compilation time
You are now compiling the full flow!!
aoc: Selected default target board a10gx
aoc: Running OpenCL parser....
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:116:16: warning: unknown attribute 'task' ignored
__attribute__((task))
               ^
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:259:5: error: expected a for, while, or do-while loop to follow '#pragma ivdep'
                                Item:for(unsigned int  win_itm_xyz=0; win_itm_xyz<item_loop_bound; win_itm_xyz++){
                                ^
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:259:5: error: expected a for, while, or do-while loop to follow '#pragma ivdep'
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:399:16: warning: unknown attribute 'task' ignored
__attribute__((task))
               ^
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:611:16: warning: unknown attribute 'task' ignored
__attribute__((task))
               ^
/home/vinay/Documents/opencl_fpga/PipeCNN/project/device/conv_pipe.cl:1026:16: error: 'max_work_group_size' attribute requires exactly 3 arguments
__attribute__((max_work_group_size(LRN_MAX_LOCAL_SIZE)))
               ^
3 warnings and 3 errors generated.
Error: OpenCL parser FAILED
Makefile:139: recipe for target 'conv.aocx' failed
make: *** [conv.aocx] Error 1

dkjiang2018, 2019-09-10:
Hi,
1.For the first and second errors, you can try to delete"Item:"
2.For the third error , you can try to modify it to "attribute((max_work_group_size(LRN_MAX_LOCAL_SIZE,1,1)))".
3.What's more, why do you  change emulator to emulator -legacy-emulator? In my knowledge, in v19.2, it doesn't need to modify, the  "-legacy-emulator" option will use in v19.3 for fast emulation.
Thanks

ignvinay, 2019-09-10:
Hi Jiang,
    I did those changes ,  thanks for your reply.
Intel has advised me to use legacy-emulator, and as per my knowledge its part of 19.2 (not 19.3).

closed 2019-09-13 by ignvinay

doonny, 2019-09-19:
@dkjiang2018  You get a new verion 19.3 after 19.2 ?

dkjiang2018, 2019-09-19:
@doonny yes, But v19.3 is currently only used internally for beta testing and it will be released in the Q4.