doonny/PipeCNN #123: POOL_LBUF_DEPTH for Xilinx Device

WenyeLiu, 2019-12-03:
Hi, I am trying to build PipeCNN on Xilinx Device. However, the parameter POOL_LBUF_DEPTH required by conv_pipe_xilinx.cl has not defined in hw_param.cl. Any suggestion value for this parameter?

WenyeLiu, 2019-12-04:
I have replaced the POOL_LBUF_DEPTH with POOL_MAX_SIZE and generated the conv.xclbin successfully. While I meet the following error when executing ./run.exe conv.xclbin ,

ERROR: Invalid scalar argument size, expected 4 got 1
ERROR: clSetKernelArg() for kernel "maxPool", argument index 0.

ERROR: CL_INVALID_ARG_SIZE
Location: host/main.cpp:898
Failed to set argument 0 of kernel pool

I am using the Xilinx Alevo U200 and SDAccel 2018.2

Any suggestion on this error?

cococener, 2019-12-16:
> I have replaced the POOL_LBUF_DEPTH with POOL_MAX_SIZE and generated the conv.xclbin successfully. While I meet the following error when executing ./run.exe conv.xclbin ,
> ERROR: Invalid scalar [...]

I want to know how you build it successfully?Because I build it failed,and the error is Unknown extension 'cl_intel_channels' - ignoring [-Wignored -pragmas];Unknown type name 'channel' .Then xilinx workers tells me that isn't a xilinx program,so the error occurs!Can you tell me how did you do?

nazarihome, 2020-05-12:
@WenyeLiu and @cococener I was wondering if you could build it successfully for Xilinx device. I have been trying for quite some time to build it for U200 and no success yet.