doonny/PipeCNN #126: Need to calculate frame rate

hamsikamahale, 2020-01-13:
Hi,
     I want to calculate frame rate. There are two timings get displayed. One is Kernel time and the other one is inference time. Which time I should consider to calculate frame rate?

doonny, 2020-01-16:
From the system aspect, inference time should be used if you have not implemented a pipeline to overlap these operations.

hamsikamahale, 2020-01-16:
Thanks a lot for your reply and information.

hamsikamahale, 2020-01-20:
Hi doony,
    We are actually implementing ResNet101 using the same code base. While displaying "Total kernel time", the time taken for eltwise kernel is not considered. Should the eltwise kernel time be considered in computation of frame rate? (want to consider only kernels and not host)

   We have tried combinations of LANE_NUM and VEC_SIZE for better frame rate. What are other options those can be tried for the same?