doonny/PipeCNN #131: RTL synthesize BUG about DSP and LUT

tommygyl, 2020-04-10:
Dear Mr.Wang，
Why when I synthesize RTL（VerilogHDL）, multiplication cannot be realized by DSP but by LUT?
If I change the MAC in pipeCNN by  * (mulitiply operation)  in OpenCL, DSPs will be synthesized.
My guess is that the compiler cannot recognize the * operation in Verilog, how can I correct this error？
Thank you!

doonny, 2020-04-12:
How do you find that DSP is synthesized into LUTs ?

tommygyl, 2020-04-13:
Sorry, it is logic utilization not LUTs. When I synthesize the piprCNN, the DSPs usage is only 3％.
board：Intel® Programmable Acceleration Card with Intel® Arria® 10 FPGA GX
SDK：Intel FPGA SDK for OpenCL 17.1
[image]

tommygyl, 2020-04-13:
I changed the RTL logic by simple multiply operation. It still could not use DSPs.

doonny, 2020-04-13:
Please check the resource utilization in the aocx file. This report is not accurate since we did not specific the DSP usage of the RTL module in the rtl.html file

tommygyl, 2020-04-13:
I understand what you mean, but how can we get resource utilization with aocx file? It seems that aocx can't be opened by any software.

sergio14890, 2020-04-15:
Hey tommy and doonny!
I have a de1soc and I would like to try pipeCNN because I think it's spectacular!
Can you tell me what procedures should I do? I honestly didn't get it very well.
If someone could send me 1 contact I would love it

doonny, 2020-04-17:
Any text editor can open the aocx file

sergio14890, 2020-04-17:
Tommygyl How did you manage to use pypeCNN one bsp17.1?